platformio/platform-espressif32 #1275: espressif32@6.0.1 Not working for board_build.partitions = default_16MB.csv

demirschmid61, 2024-01-07:
Hello Team,

Thank you very much for the upgrade of the espressif32 version with 6.0.1.

I want to use the **board_build.partitions = default_16MB.csv** for ESP32 but controller is reseting. If I am using **board_build.partitions = huge_app.csv** its working fine.

I am using Platformio, is there any other **board_build.partitions** i need to select?

The **platform.ini** file is mentioned below.

`[env:esp32dev]
platform = espressif32@6.0.1
board = esp32dev
framework = arduino
monitor_speed = 115200

board_build.partitions = huge_app.csv         //Working fine for huge_app partition
; board_build.partitions = default_16MB.csv //Resetting when default_16MB used
`

/***************************Reset Logs***************************/
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
ets Jul 29 2019 12:21:46
/**********************************************************************/

Jason2866, 2024-01-07:
The partition scheme you use needs to align with the size of the flash chip mounted on the used board.

valeros, 2024-01-08:
Hi @demirschmid61, what board are you using? Does it really have 16MB of flash?

closed 2025-02-25 by stale[bot]