platformio/platform-espressif32 #1276: Feature Request:board config for ESP32-C3FN4

jonnytest1, 2024-01-09:
- [x ] **Feature Request**.
     missing board https://www.waveshare.com/esp32-c3-zero.htm

     freq:160MHz
     4MB of onboard Flash
     400KB of SRAM and 384KB ROM (not sure what this eans exactly with the 4MB rom flash ^^)

valeros, 2024-01-09:
It seems you can get around with a similar board, for example `esp32-c3-devkitc-02`.

jonnytest1, 2024-01-09:
thats what i did so far yes , but it seems to have "unique" sizes for ram which could potentially mean some firmwares could be blocked when they shouldnt (or the other way round)

jonnytest1, 2024-01-09:
[image]

did i miss something ?

Jason2866, 2024-01-09:
> thats what i did so far yes , but it seems to have "unique" sizes for ram which could potentially mean some firmwares could be blocked when they shouldnt (or the other way round)

RAM is same for all C3 MCUs. If you find different settings it mentions overall RAM against useable one.

valeros, 2024-01-09:
@jonnytest1 

> did i miss something ?

You're probably using an old platform version, just run `pio pkg update -g -p espressif32` in the IDE terminal to install the latest version v6.5.0.

closed 2024-01-09 by jonnytest1