platformio/platform-espressif32 #1278: Open DTU on Battery brings Tasmota compiling apperently in trouble.

Solar320, 2024-01-14:
current state: I have PlattformIO and VisualStudio code installed.
I have currently 3 Projekts opened with PlattformIO and all 3 are in the Workspace.
The 3 Projects are Open DTU on Battery, Daly2MQTT and Tasmota.

If I want to compile Tasmota I get errors like this:

**Retrieved `.pio\build\tasmota32-core2\partitions.bin' from cache
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directoryRetrieved `.pio\build\tasmota32-core2\lib714\freetype\bdf\bdf.c.o' from cache
*** [.pio\build\tasmota32-core2\src\tasmota.ino.cpp.o] Error 1
============================================================= [FAILED] Took 32.44 seconds =============================================================

Environment      Status    Duration
---------------  --------  ------------
tasmota32-core2  FAILED    00:00:32.441**

The solution of this should be to delete the .PlattformIO folder at users\....... an wait feeled 2 hours to coming to work again.

I found out the following:

If I compile the Open DTU on Battery Project only one time (ESP32 variant)  it will download additional things into the /packages folder
named : framework-arduinoespressif32@src-fa0e0f7300c07ae80d2159cce942c530  and 
tool-esptoolpy@src-7942a1fc65a64a7a2d720e5d6b65f4b3

As long this 2 folders exist in the Packages folder, the Tasmota compiling errors appear.

Solution for me was to copy this 2 folders to an Backup folder like "OPEN-DTU-BACKUP"
As soon as this 2 folders disappear from the User\username\.plattformIO\packages folder, you can compile Tasmota without any errors.
If you want to compile an open DTU Projekt again you have to copy this 2 removed folders back to the packages folder or you  have to wait until the project download and create the 2 folders again.   
I also compiled the daly2mqtt projekt, it also download and create e.g. "tool-mklittlefs@1.203.210628" and others. The difference is that they will not interfer with tasmota and will not disturb the compiling process.

The only visible difference in the name I see, is that Open DTU on battery named his folders with "@src-" and daly2Mqtt named it "@1.xxxx"
May be this combination @src- is the problem? Normaly "SRC" is reserved for the Main directory were the *.ino or Main.cpp files are in, may be this "SRC"  should not be used in the packages folder.  I am a beginner with this stuff and I am not sure if you can reproduce this at your PC.  I use Win11 if this is relevant.

Jason2866, 2024-01-14:
@Solar320 Thx for the detailed steps what you did to get this error. I have now an idea what's happening. Tasmota is using the original espressif framework when an project is build which uses `platform = espressif32 @ 6.5.0`. Have to find a way to solve this. Stay tuned.
To be clear it is not a bug of the Platformio Platform. It is a not supported use case.

Jason2866, 2024-01-14:
@Solar320 Do not run in this error. Habe both projects (Tasmota / Open DTU on Battery) in one Workspace.
Both compile without any error. Can switch between. The issue i thought, does not happen. Always the correct framework is used. Tbh, i am clueless whats causing your issue

Jason2866, 2024-01-14:
Mhh, could it be you are not using latest Tasmota? Because the env:tasmota32-core2 does not build anymore with actual latest version!
You need to have in file `platformio_tasmota32.ini` this platform settings:
```
[core32]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.00/platform-espressif32.zip
platform_packages           =
build_unflags               = ${esp32_defaults.build_unflags}
build_flags                 = ${esp32_defaults.build_flags}

[core32_30]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.10/platform-espressif32.zip
platform_packages           =
build_unflags               = ${core32.build_unflags}
build_flags                 = ${core32.build_flags}
```

Solar320, 2024-01-14:
Ok I download now the tasmota 13.3 prerelease which has the old 2023 entries in its platformio_tasmota32.ini

[core32]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2023.11.01/platform-espressif32.zip
platform_packages           =
build_unflags               = ${esp32_defaults.build_unflags}
build_flags                 = ${esp32_defaults.build_flags}

[core32_30]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2023.12.10/platform-espressif32.zip
platform_packages           =
build_unflags               = ${core32.build_unflags}
build_flags                 = ${core32.build_flags}

And I download the Tasmota development:

[core32]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.00/platform-espressif32.zip
platform_packages           =
build_unflags               = ${esp32_defaults.build_unflags}
build_flags                 = ${esp32_defaults.build_flags}

[core32_30]
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/2024.01.10/platform-espressif32.zip
platform_packages           =
build_unflags               = ${core32.build_unflags}
build_flags                 = ${core32.build_flags}

It has the new Date 2024 inside. 

I compiled the developer version as ESP8266, OK
Then I compiled as developer Tasmota32-core2 
Is this the right source?
https://github.com/arendst/Tasmota/tree/development
hours later..........
OK. no errors.

Now I switch to Open DTU on Batteries and compile, compile.... ok, then switch back to tasmota32-core2 and compile again.
[image]

NNNNOOOOOOOO again compiler errors!
I also try to compile the normal tasmota32 development, still the same error.
When I delete all folders (inside "packages") with "SCR" in his name, the errors are gone. (in the last picture below you can see which folders I deleted in the packages folder to get the compiler working with the developer version, these files are coming from OPEN DTU on Battery)

Now an test with 13.3 prerelease: Tasmota32

And?......OK!?!
prerelase 13.3  Tasmota32-core2...also .......OK!

The current state is now:

Install all the projekts: compile once Open DTU on Battery (ESP32 one) , it will create the files similar to "tool-esptoolpy@src-dc58dc5b29d0330f1a2c51773f63e2a2" into the
Packages folder.

After that I switch to Tasmota32 Developer (both 2024 version) or Tasmota32-core 2, the result are these compiling errors.
(They only disappear if I delete the files from Open DTU on Battery with "@SCR-XXXX" in his name.)

But in the same situation, If I compile with Tasmota32 13.3 Prerelease or Tasmota32-core2 Prerelease 13.3 there are no compiler errors! The prerelease version is in the same workspace.
[image]

Since I had also the old developmentversion from last year before and it has the same errors it has something to do with the development version only.

**So I will now use only the 13.3 Prerelease for now , to prevent to waste your time. May be this error will disappear in the future may be not.**
The Problem is not explainable......

I may try to reproduce this error also on an other System with windows 10, but this Intel i5 thing is very very old and slow.

At last the picture of the files I delete to get the developer version running.
[image]

closed 2025-02-25 by stale[bot]