platformio/platform-espressif32 #1280: wanted to see if there is any support for ESP32-S3-WROOM-2-N32R8V

prschguy1, 2024-01-14:
While I see there is some support for the 16mb versions of this board, not seeing anything current for the 32mb version of it. I am using it on a couple of different builds, and I am running out of memory. Doesn't seem like I should be. Have some esp32 v1 boards that out perform this chip, and seems that shouldn't be happening. While I know that the authentic espressif boards should have memory set as opi_opi, and the generic versions of that are using quad memory. Was hoping someone here might have a board define jason that works with platformio, as well as the appropriate memory table that would go with. I know it's a big ask, but this one has been out for a while now, and from what I read, many have struggled with it. Hope someone can help.

Thanks, Craig

Jason2866, 2024-01-15:
This should Do

```
{
  "build": {
    "arduino":{
      "ldscript": "esp32s3_out.ld",
      "memory_type": "opi_opi"
    },
    "boot": "opi",
    "core": "esp32",
    "extra_flags": [
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1",
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_USB_MODE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "dout",
    "hwids": [
      [
        "0X303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "ESP32S3 32MB Flash 8MB PSRAM OPI/OPI",
  "upload": {
    "flash_size": "32MB",
    "maximum_ram_size": 327680,
    "maximum_size": 33554432,
    "require_upload_port": true,
    "speed": 921600
  },
  "url": "https://www.espressif.com/sites/default/files/documentation/esp32-s3-wroom-2_datasheet_en.pdf",
  "vendor": "Espressif"
}
```

prschguy1, 2024-01-15:
Thank you so much for that. Will give it a go and see if it helps the memory issue.

Craig

MoonLight423, 2024-01-25:
I use N16R8 with this "platformio.ini"
`[env:esp32-s3-devkitc-1]
platform = espressif32
framework = arduino
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.filesystem = fatfs
board_build.arduino.memory_type = qio_opi
board_build.arduino.partitions = app3M_fat9M_16MB.csv
board_upload.flash_size = 16MB
monitor_speed = 115200
lib_ldf_mode = deep+
build_flags = 
	-Iinclude/
	-DCORE_DEBUG_LEVEL=5
	-DCONFIG_ARDUHAL_LOG_COLORS=1
	-DBOARD_HAS_PSRAM=1
	-mfix-esp32-psram-cache-issue
	-DLV_LVGL_H_INCLUDE_SIMPLE=1
	-DLV_CONF_INCLUDE_SIMPLE=1`

19e2k67, 2024-06-04:
The model of my development board is "ESP32-S3-WROOM-1-N16R8", which is set as follows in Platformio:
```c
[env:esp32s3]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
; 指定为16MB的FLASH分区表
board_build.arduino.partitions = default_16MB.csv
; 指定FLASH和PSRAM的运行模式
board_build.arduino.memory_type = qio_opi
; 预定义宏，启用PSRAM
build_flags = -DBOARD_HAS_PSRAM
; 指定FLASH容量为16MB
board_upload.flash_size = 16MB
```
Reference link: [Reference link](https://www.cnblogs.com/macrored/p/17357581.html)

19e2k67, 2024-06-04:
ESP32-S3-N32R8:
板子选择 Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD,No PSRAM) ，再到platformio.ini文件下修改配置。配置如下：
   ```C
  [env:esp32-s3-devkitc-1]
  platform = espressif32
  board = esp32-s3-devkitc-1
  framework = arduino
  lib_ldf_mode = deep+	#在搜索库依赖关系时使用 "深度+" 模式
  board_build.arduino.partitions = default_16MB.csv	
  board_build.arduino.memory_type = opi_opi	#8MB PSRAM的运行模式
  build_flags = -DBOARD_HAS_PSRAM		#用于指示编译器当前的目标设备是否具有 PSRAM
   -mfix-esp32-psram-cache-issue #用于修复 ESP32 上的 PSRAM 缓存一致性问题的编译选项
  board_upload.flash_size = 32MB #FLASH大小
  #添加的库
  lib_deps = 
   bblanchon/ArduinoJson@^6.21.2
   lvgl/lvgl@^8.3.7
   bodmer/TFT_eSPI@^2.5.30
   lorol/LittleFS_esp32@^1.0.6
```
其中最主要的就是“board_build.arduino.memory_type = opi_opi”，查看官方文档可以发现ESP32的PSRAM的运行模式是有Quad SPI和Octal SPI的区别的，而8M的PSRAM使用的是Octal SPI，默认的Quad SPI是运行不了的。
原文链接：https://blog.csdn.net/kzc1209/article/details/131111920