platformio/platform-espressif32 #1282: preserve_source_file_extension Causing inability to load

Kreedb, 2024-01-16:
When I was using version idf-4.4.5, the option "board_build. esp idf. reserves source file extension=yes" worked fine

After switching to IDF-4.4.6 version, I was unable to load the program and kept getting stuck in this place

I (25) boot: ESP-IDF 4.4.6 2nd stage bootloader
I (25) boot: compile time 11:29:44
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0ade4h ( 44516) map
I (105) esp_image: segment 1: paddr=0001ae0c vaddr=3fc91ba0 size=02050h (  8272) load
I (107) esp_image: segment 2: paddr=0001ce64 vaddr=40374000 size=031b4h ( 12724) load
I (116) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1fce4h (130276) map
I (145) esp_image: segment 4: paddr=0003fd0c vaddr=403771b4 size=0a9ech ( 43500) load
I (161) boot: Loaded app from partition at offset 0x10000
I (161) boot: Disabling RNG early entropy source...

Attached is my testing project
[espidf-arduino-blink.zip](https://github.com/platformio/platform-espressif32/files/13945039/espidf-arduino-blink.zip)

Kreedb, 2024-01-16:
I have confirmed the commit that caused this issue

#https://github.com/espressif/esp-idf/commit/2183fbc046d37eef47b5b3d336f80f0fc4bb8170

Jason2866, 2024-01-16:
What does the option? With a fast search I didn't found any hint.

Kreedb, 2024-01-16:
board_build.esp-idf.preserve_source_file_extension = yes

I'm very sorry that the code was transcoded by GitHub

Annotate this line and it will run normally
https://github.com/espressif/esp-idf/blob/2183fbc046d37eef47b5b3d336f80f0fc4bb8170/components/esp_system/port/cpu_start.c#L355
I have identified the problem, but I don't understand why this approach is feasible
Their code seems completely unrelated

valeros, 2024-01-17:
> When I was using version idf-4.4.5, the option "board_build. esp idf. reserves source file extension=yes" worked fine
After switching to IDF-4.4.6 version, I was unable to load the program and kept getting stuck in this place

I'm not sure that the `preserve_source_file_extension` option has something to do with your problem as it's related to the build process not the runtime of your application, probably something changed in v4.4.6 that broke your project.

Kreedb, 2024-01-17:
But once I turn off this option, the 4.4.6 version works normally
I also feel that this option should not be the reason, but I cannot find the cause of the error.
Can you try the test project I sent?  It will reproduce this bug.

Jason2866, 2024-01-17:
Still don't know why using `preserve_source_file_extension`? What should this option do?
Never needed this option. Have you a link to espressif docu for?  Do not expect that something found in source code which is not documented is working in a newer release.

Kreedb, 2024-01-18:
> Still don't know why using `preserve_source_file_extension`? What should this option do? Never needed this option. Have you a link to espressif docu for? Do not expect that something found in source code which is not documented is working in a newer release.

https://github.com/espressif/esp-dsp/tree/master/modules/support/cplx_gen
Because I used the esp-dsp component
If I do not use this option, the file names compiled from dsps_cplx_gen.S and dsps_cplx_gen.c are both dsps_cplx_gen.o, which will fail to compile

valeros, 2024-01-23:
Hi @Kreedb, thanks for reporting. Indeed, there is an issue with the final binary image when the `preserve_source_file_extension` option is enabled. In a nutshell, the `ldgen.py` script responsible for generating one of the linker scripts simply [doesn't take into account](https://github.com/espressif/esp-idf/blob/master/tools/ldgen/ldgen/entity.py#L184-L186) object files with the following pattern `.*.o`. I've temporarily patched our IDF packages with a hotfix, so please update your platform with the following command `pio package update -g -p espressif32`.

Kreedb, 2024-01-23:
> Hi @Kreedb, thanks for reporting. Indeed, there is an issue with the final binary image when the `preserve_source_file_extension` option is enabled. In a nutshell, the `ldgen.py` script responsible [...]

Thank you, but when using esp-idf in conjunction with Arduino, the idf version can only use 4.4.6, and I did not find any fix measures in this version of the code

valeros, 2024-01-23:
I've updated both packages the latest v5.1.2 and 4.4.6 for Arduino, so updating the platform itself via the command I mentioned above should be enough as it should trigger package updates as well. Alternatively, you can just delete the `$HOME/.platformio/packages/framework-espidf*` folder and rerun your project.

Kreedb, 2024-01-23:
It's amazing. It's working fine now. Can you know where you modified it to solve this bug

Kreedb, 2024-01-23:
Thank you very much. I have found the parts you modified and posted them for future reference
[image]

closed 2024-01-23 by Kreedb

Jason2866, 2024-01-23:
For IDF 4.4 the change has to be added here https://github.com/espressif/esp-idf/blob/release/v4.4/tools/ldgen/entity.py#L195

Kreedb, 2024-05-29:
@valeros hey, 4.4.7 idf Forgot this patch