platformio/platform-espressif32 #1283: ESP32-S3 invalid header

Gheb17, 2024-01-16:
I'm trying to program this [ESP32-S3 based board](https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28), but in thonny i receive this message:
invalid heade ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x38 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049b21
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff

Gheb17, 2024-01-16:
> Your flash attempt(s) failed. The flash is empty.

but the upload of firmware finish successfully:
"C:\Program Files (x86)\Thonny\python.exe" -u -m esptool --port COM10 erase_flash
esptool.py v4.4
Serial port COM10
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:27:e2:3d:c3:18
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 42.8s
Hard resetting via RTS pin...
Erasing done
------------------------------------

"C:\Program Files (x86)\Thonny\python.exe" -u -m esptool --port COM10 write_flash --flash_mode keep --flash_size detect 0x1000 "D:/backup/progetti/Progetti in corso/Spada cyberpunk/Esp32-s3-touch-lcd-1.28-code/S3-Touch-LCD-1.28-MPY.bin"
esptool.py v4.4
Serial port COM10
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 48:27:e2:3d:c3:18
Stub is already running. No upload is necessary.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00001000 to 0x00183fff...
Compressed 1583792 bytes to 1050433...
Writing at 0x00001000... (1 %)
Writing at 0x00013164... (3 %)
Writing at 0x0001b2ed... (4 %)
Writing at 0x00022660... (6 %)
Writing at 0x00028fc3... (7 %)
Writing at 0x00034342... (9 %)
Writing at 0x0003d1bf... (10 %)
Writing at 0x000424f9... (12 %)
Writing at 0x0004c0b0... (13 %)
Writing at 0x00055312... (15 %)
Writing at 0x0005bb71... (16 %)
Writing at 0x000613d4... (18 %)
Writing at 0x00066699... (20 %)
Writing at 0x0006b6e0... (21 %)
Writing at 0x00070794... (23 %)
Writing at 0x00075a03... (24 %)
Writing at 0x0007a4f4... (26 %)
Writing at 0x0007f2a5... (27 %)
Writing at 0x000854d6... (29 %)
Writing at 0x0008b19f... (30 %)
Writing at 0x00091109... (32 %)
Writing at 0x000963b8... (33 %)
Writing at 0x0009b383... (35 %)
Writing at 0x000a0923... (36 %)
Writing at 0x000a66a0... (38 %)
Writing at 0x000aba48... (40 %)
Writing at 0x000b1053... (41 %)
Writing at 0x000b66f6... (43 %)
Writing at 0x000bbf1c... (44 %)
Writing at 0x000c1439... (46 %)
Writing at 0x000c64ee... (47 %)
Writing at 0x000cb9bd... (49 %)
Writing at 0x000d167b... (50 %)
Writing at 0x000d71ed... (52 %)
Writing at 0x000dd4ad... (53 %)
Writing at 0x000e2340... (55 %)
Writing at 0x000e72e2... (56 %)
Writing at 0x000ec490... (58 %)
Writing at 0x000f1810... (60 %)
Writing at 0x000f6c56... (61 %)
Writing at 0x000fc0a7... (63 %)
Writing at 0x00101258... (64 %)
Writing at 0x00106295... (66 %)
Writing at 0x0010b249... (67 %)
Writing at 0x00110668... (69 %)
Writing at 0x00115bd0... (70 %)
Writing at 0x0011b252... (72 %)
Writing at 0x00120466... (73 %)
Writing at 0x00125f24... (75 %)
Writing at 0x0012ae65... (76 %)
Writing at 0x00130092... (78 %)
Writing at 0x00134ee5... (80 %)
Writing at 0x0013a3c5... (81 %)
Writing at 0x0013f695... (83 %)
Writing at 0x00144df8... (84 %)
Writing at 0x0014a67e... (86 %)
Writing at 0x00150230... (87 %)
Writing at 0x0015678f... (89 %)
Writing at 0x00160897... (90 %)
Writing at 0x00166a4e... (92 %)
Writing at 0x0016bd8f... (93 %)
Writing at 0x001717c5... (95 %)
Writing at 0x00176bd3... (96 %)
Writing at 0x0017c5e8... (98 %)
Writing at 0x00182733... (100 %)
Wrote 1583792 bytes (1050433 compressed) at 0x00001000 in 92.7 seconds (effective 136.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done!

Kreedb, 2024-01-17:
you bootloader is error , try to use platfomrio upload

Kreedb, 2024-01-17:
"bootloader" : { "offset" : "0x0", "file" : "bootloader.bin", "encrypted" : "false" },
    "app" : { "offset" : "0x10000", "file" : "firmware.bin", "encrypted" : "false" },
    "partition-table" : { "offset" : "0x8000", "file" : "partition-table.bin", "encrypted" : "false" },
    
    this file in platformio build_dir

Jason2866, 2024-01-17:
The issue is not related to Platformio. When building a firmware and flashing with Platformio, Platformio takes care to flash all needed at correct places.
As mentioned earlier your flash command lacks mandantory files. Without the device will never boot successfully.

Jason2866, 2024-01-17:
Google search ;-)  Random hit https://www.martinloren.com/how-to/fashing-esp32/

Gheb17, 2024-01-17:
> > The issue is not related to Platformio. When building a firmware and flashing with Platformio, Platformio takes care to flash all needed at correct places. As mentioned earlier your flash command lacks mandantory files. Without the device will never boot successfully.
> 
> Ok, there are any guide o tutorial? I'm sorry but i'm new with Platformio and visual studio

> Google search ;-) Random hit https://www.martinloren.com/how-to/fashing-esp32/

thanks, but now i don't know where find the bootloader.bin and partition-table.bin, I tried to research without success, for firmware i think I should use the firmware I have already tried to upload

Gheb17, 2024-01-17:
> Search all your disks, you will find it

i think i find it, in the exaple folder provider by waveshare, but I still have the problem with thonny
[image]

1 kill
test offset :  4096 0x1000
case ok
D:\backup\progetti\Progetti in corso\Spada cyberpunk\Esp32-s3-touch-lcd-1.28-code
C:\Users\gabri\OneDrive\Desktop\bootloader.bin
C:\Users\gabri\OneDrive\Desktop
C:\Users\gabri\OneDrive\Desktop\partitions.bin
C:\Users\gabri\OneDrive\Desktop
D:\backup\progetti\Progetti in corso\Spada cyberpunk\Esp32-s3-touch-lcd-1.28-code\S3-Touch-LCD-1.28-MPY.bin
D:\backup\progetti\Progetti in corso\Spada cyberpunk\Esp32-s3-touch-lcd-1.28-code
D:\backup\progetti\Progetti in corso\Spada cyberpunk\Esp32-s3-touch-lcd-1.28-code
C:\Users\gabri\OneDrive\Desktop\partitions.bin
2 kill
3 kill
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
test offset :  4096 0x1000
case ok
test offset :  32768 0x8000
case ok
test offset :  65536 0x10000
case ok
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
NO XMC flash  detected!
SPI_BOOT_CRYPT_CNT 0
SECURE_BOOT_EN False
Compressed 15104 bytes to 10401...
Compressed 3072 bytes to 146...
Compressed 1583792 bytes to 1050433...

 is stub and send flash finish

valeros, 2024-01-17:
Hi there, the issue tracker in this repository is not the right place for questions not related to PlatformIO.

closed 2024-01-17 by valeros