platformio/platform-espressif32 #1285: WifiAP fails on ESP32 with 6.5.0, works with 6.1.0

scudderfish, 2024-01-17:
Using an ESP32 as a soft AP fails under 6.5.0.  Clients will refuse to connect after supplying the password.  The same code compiled with 6.1.0 works.

```
#include <Arduino.h>
#include <WiFi.h>

IPAddress ap_local_IP(172, 16, 17, 1);
IPAddress ap_gateway(172, 16, 17, 1);
IPAddress ap_subnet(255, 255, 255, 0);

const char *ssid = "TestSSID";
const char *password = "password";

  void setup() {

    Serial.begin(115200);

    Serial.println("Failed to find the master, upgrading");
    Serial.println(WiFi.softAP(ssid, password) ? "soft-AP setup" : "Failed to connect");
    delay(2000);
    Serial.println(WiFi.softAPConfig(ap_local_IP, ap_gateway, ap_subnet) ? "Configuring Soft AP" : "Error in Configuration");
    Serial.println(WiFi.softAPIP());
  
   }

void loop() {
}
```

```
[env:esp32dev]
; FAILS platform = espressif32@6.5.0
platform = espressif32@6.1.0
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_filters = esp32_exception_decoder
```

Jason2866, 2024-01-17:
Does the example work with ArduinoIDE Core 2.0.14? If no, it is not a Platformio related issue.

scudderfish, 2024-01-20:
Downloaded latest Arduino IDE (2.2.1) and the sample works fine there.

Jason2866, 2024-01-20:
ArduinoIDE is just the Platform used for compiling a project. It does not tell which Arduino core version was used.

scudderfish, 2024-01-20:
Ah, not used to how the Arduino platform works :) If I do a verbose compile I see `.arduino15/packages/esp32/hardware/esp32/2.0.11` so I assume that is behind the version you're after.  Now to find out how to change that....

scudderfish, 2024-01-20:
OK, figured out how to do that, and with 2.0.14 I cannot connect to the SoftAP, it is reproduced in Arduino Code 2.0.14 so not a platformio specific issue.  Closing this as it is probably the things you linked to.