platformio/platform-espressif32 #1288: Need help with a build flag

meakashrao, 2024-01-25:
For an application of mine, i need to write to a protected flash location. 

I am trying to erase the flash sector (offset: 0x8000 and size 0x1000 which is for my partition table space), I am using ESP32-Wroom-32E. It is said that although the length of the partition table is 0xC00 bytes but due to checksum it ends up occupying the whole flash sector, I have verified the offset to be 0x8000 from the sdkconfig.h
#define CONFIG_PARTITION_TABLE_OFFSET 0x8000

But while erasing the flash sector, I get an error E (16696) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[image]

[image]
To achieve this, I used the flag, 
[image]

I am not using coredump, nonethless, I erased the whole flash, did serial coding with min_spiffs.csv with coredump as the partition and then tried to update the partition but I still get the same error.

Am i using the correct flags. PLease help

Jason2866, 2024-01-25:
@meakashrao  This is a compile time IDF option. You cant change in Arduino! The Arduino libs needs to be compiled with the option set!

meakashrao, 2024-01-25:
I also tried changing the status of the flag in sdkconfig, I know that sdkconfig is generated and we are not supposed to edit it, but if I edit it, should it not allow? any official comments. 

And I still don't understand why isn't it configurable?. Thank you for your help on this one @Jason2866

Jason2866, 2024-01-25:
sdkconfig is changeable for IDF projects. When using Arduino, precompiled libs are used in Arduino projects. In this libs most options set in sdkconfig can not be changed since "burned" in the libs. If you want full control off every option use Arduino as a component of IDF.

meakashrao, 2024-01-25:
> If you want full control off every option use Arduino as a component of IDF.

Oh, OK thank you

valeros, 2024-01-29:
Closing as the IDF's compile-time configuration settings cannot be applied to the Arduino framework.

closed 2024-01-29 by valeros