platformio/platform-espressif32 #1305: PIO Inspect does not work.

zekageri, 2024-02-09:
Hi! I wanted to analyse my codebase with Inspect.

I had an error at the first inspection which said that my program is bigger than it can be.
Which is strange because i can upload it fine.
After some research i found out that it built the project in debug mode therefore it is a lot bigger.
So i added a line into my ini file to optimise the debug build. 

`debug_build_flags = -Os`

Now if i Inspect it again, it does not produce an error but stuck at 100% and it seems like it will never finish.

I'm using an ESP-wrover-E with 16mb flash and 8mb psram.
I'm using a lot of malloc in my project to fully use the external ram as i can.

What can i do to make the Inspection work?

I will also try the following build flags in the ini file after i waited long enough for the Inspection to finish... ( I suspect it never will )

```cpp
build_flags =
-DBOARD_HAS_PSRAM
-mfix-esp32-psram-cache-issue
-DCONFIG_MBEDTLS_DYNAMIC_BUFFER=1
-DCONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=1
-DCONFIG_SPIRAM_CACHE_WORKAROUND=1
```

**Here is my current ini file**

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp-wrover-kit]
platform = espressif32

board       = esp-wrover-kit
framework   = arduino
board_build.filesystem = littlefs

board_build.f_cpu   = 240000000L 
upload_protocol = espota
upload_port     = esp
upload_speed    = 921600
monitor_speed   = 115200
monitor_filters = colorize, esp32_exception_decoder

board_upload.flash_size = 16MB
board_build.flash_mode = qio
board_build.partitions = ./Partitions/hshPartition_APP_3MB.csv
board_build.f_flash    = 80000000L 

debug_build_flags = -Os

build_flags =   
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
    -mfix-esp32-psram-cache-strategy=memw
    -DCORE_DEBUG_LEVEL=0
    -D ARDUINOTRACE_ENABLE=0
    -std=gnu++17
    -DCONFIG_ARDUINO_ISR_IRAM=0
    -DCONFIG_SPIRAM_USE_MALLOC=1
build_unflags =
    -std=gnu++11

upload_flags = 
    --auth=aSuperSecretUploadPassword
```

valeros, 2024-02-12:
Hi @zekageri, any chance you could share your project to reproduce the issue?

zekageri, 2024-02-12:
Unfortunately i can't and it also is a massive app. I can try to build a small project and hope it triggers it.

valeros, 2024-02-12:
> I can try to build a small project and hope it triggers it.

Would be great.

zekageri, 2024-02-16:
Well, i was able to inspect only the RAM usage

[image]

But this is obviously not true.
I had to add this flag `debug_build_flags = -Os` to my ini file and i had to uncheck the "Check Code" toggle.
If it is checked it will never finish. I will give it another try. Will let it sit for hours and see what it will do.

**Just some questions**

Does PIO inspect sees the external ram? My ini file contains the board definition and some EXT ram build flags.
Does PIO calculate malloc and new when inspecting?

valeros, 2024-02-16:
> Will let it sit for hours and see what it will do.

It shouldn't take that long. How many files does your project contain?

> Does PIO inspect sees the external ram?

PIO analyzes your firmware section by section. In a nutshell, if a symbol is located in a section that occupies memory during app execution then it will considered as RAM.

> Does PIO calculate malloc and new when inspecting?

Unfortunately, it doesn't. Unlike the static memory allocation, the dynamically allocated memory (in your case the memory allocated via `malloc`) can vary based on application requirements at the runtime, so a more complex instruments like Valgrind are required.

zekageri, 2024-02-16:
> It shouldn't take that long. How many files does your project contain?

Well, I don't know. A lot. A rough estimate is (37 * 2) 74 if each folder contains at least two files. But some files are more than 3-4 files so give or take.

Thank you for the info. I will try to create a smaller project and see what it does.

The `unknown` 217 kb file is really strange to me btw... It is probably some Espressif content.

closed 2025-04-25 by stale[bot]