platformio/platform-espressif32 #1306: Add support for "ESP32-C3 SuperMini" board

barsikus007, 2024-02-12:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Information

- <https://microsoft.github.io/devicescript/devices/esp32/esp32c3-supermini>
- <https://aliexpress.com/item/1005006371542813.html?sku_id=12000036934095950>

valeros, 2024-02-12:
Hi @barsikus007, there is no official variant for this board, so a proper variant header should be added to this repo https://github.com/espressif/arduino-esp32 in the first place.

sparky8512, 2024-02-13:
Looks like it already exists as variants `makergo_c3_supermini` and `nologo_esp32c3_super_mini`. Not sure if Maker go was the original designer and the "nologo" one is for clones/resellers, or what. They look essentially the same to me.

barsikus007, 2024-02-13:
> Looks like it already exists as variants `makergo_c3_supermini` and `nologo_esp32c3_super_mini`. ~Not sure if Maker go was the original designer and the "nologo" one is for clones/resellers, or what~ (apparently, Nologo is an actual manufacturer). They look essentially the same to me.

thanks, but I haven't these boards in platformio
how to add them?

valeros, 2024-02-13:
The boards will be added once a new stable version of the framework is released.

barsikus007, 2024-02-13:
thanks, I'll just wait for release and will use `nologo_esp32c3_super_mini` one

closed 2024-02-13 by barsikus007

faizannazir, 2024-03-09:
> Author

I have a esp32 c3 super mini board but i am not able to flash it

vbguyny, 2024-03-17:
Does anyone know when there will be an update available with support for the super mini?

valeros, 2024-03-18:
@vbguyny It will be added once a new stable release of Arduino core is published.

sparky8512, 2024-03-18:
I wouldn't count on that happening any time soon, though. Neither of the above-mentioned board variants is present in the v2.x release branch of arduino-esp32, and it looks like 3.0.0 is still in pre-release. I have no insight into the release schedule for that, but even once released, there seems to be [some uncertainty](https://github.com/platformio/platform-espressif32/issues/1225) around getting that version supported by platformio.

In the meantime, you can pretty much use any of the existing ESP32-C3 variants that don't have a separate USB serial chip on the board (I use `lolin_c3_mini`), as long as you don't use the board-specific Arduino pin macros like LED_BUILTIN, etc. And if you really need those, I think you can copy one of the `pins_arduino.h` files that got added in arduino-esp32 (from [here](https://github.com/espressif/arduino-esp32/blob/master/variants/nologo_esp32c3_super_mini/pins_arduino.h) or [here](https://github.com/espressif/arduino-esp32/blob/master/variants/makergo_c3_supermini/pins_arduino.h)) into your project somewhere in the `-I` header include path.

pablomuri, 2024-08-31:
Hi any news on this? PlatformIO is already using the esp32 v3.x, isn't it?

mhaberler, 2024-08-31:
> Hi any news on this? PlatformIO is already using the esp32 v3.x, isn't it?

no it isnt
#1225

pcace, 2025-02-08:
Hi, is there any way now to use these in plattformIO?

thanks a lot!

Michelingumman, 2025-02-25:
@pcace Im wondering this aswell!!