platformio/platform-espressif32 #1309: esp.emac: emac_esp32_transmit(229): insufficient TX buffer size

folkertvanheusden, 2024-02-13:
Hi,

With lots of network-traffic, the WT-ETH01 ESP32 module goes into a loop where it only prints "esp.emac: emac_esp32_transmit(229): insufficient TX buffer size"  and never returns to the main code.

I googled it and found this: https://github.com/espressif/esp-idf/issues/12037 to be exact: https://github.com/espressif/esp-idf/issues/12037#issuecomment-1685693130

Is this something that can be enabled in the platformio version?

regards

valeros, 2024-02-14:
Hi @folkertvanheusden, what framework are you using?

folkertvanheusden, 2024-02-14:
Hi,
This is with the Arduino framework.

valeros, 2024-02-14:
Unfortunately, there is no an easy way to enable that configuration with Arduino, mainly because the it's a compile-time option for ESP-IDF. I believe you can try to compile your application as a mixed Arduino/IDF project (with enabled `CONFIG_ETH_TRANSMIT_MUTEX`), take a look at an example [here](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan).

folkertvanheusden, 2024-02-14:
Thanks for investigating this!