platformio/platform-espressif32 #1311: Debug support for Arduino Nano Esp32

phant0mias, 2024-02-17:
Would it be possible to add debug suport for the Arduino Nano ESP32 (board = arduino_nano_esp32)

In the arduino espressif library it is included since release v2.0.12. 
https://github.com/espressif/arduino-esp32/pull/8567

As you switched to v2.0.14 in Dec 27, 2023 it might be possible

Thank you

Jason2866, 2024-02-17:
Have you tried? Since it is a esp32-s3 it should work in general.

phant0mias, 2024-02-17:
Yes I tried it. I have two differnet S3 (m5stack-cores3, esp32-s3-devkitc-1) and they work.
I configured it as 

platform = espressif32
board = arduino_nano_esp32
upload_protocol = esp-builtin
debug_tool = esp-builtin

But nothing happens. The output in the degub terminal is:
DEBUG: Current (esp-builtin) External (cmsis-dap, dfu, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
AVAILABLE: cmsis-dap, dfu, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-builtin

I also tried to get it into boot mode using 
https://docs.arduino.cc/tutorials/nano-esp32/debugging/
but no effort

Do you have any idea?

phant0mias, 2024-02-19:
dvanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.7% (used 44916 bytes from 327680 bytes)
Flash: [==        ]  20.4% (used 642593 bytes from 3145728 bytes)
Building .pio\build\omote_esp32\firmware.bin
esptool.py v4.5.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, dfu, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-builtin
Uploading .pio\build\omote_esp32\firmware.bin
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Error: esp_usb_jtag: could not find or open device!
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1

valeros, 2024-02-19:
> could not find or open device!

The error message is quite descriptive, you sure all necessary drivers are installed for this board?

phant0mias, 2024-02-19:
Hi,
thanks for your answer. 

I have arduino 2.x IDE installed. Following the instructions in
https://docs.arduino.cc/tutorials/nano-esp32/debugging/
I got the same error.
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 50005 for tcl connections
Info : Listening on port 50006 for telnet connections
Error: esp_usb_jtag: could not find or open device!

First I try to get it working in the arduino IDE (as it is an arduino product) and then I switch to platformio.....

phant0mias, 2024-02-20:
I am in contact with arduino and will write when my problem is found

valeros, 2024-02-21:
Closing as the issue is not related to PlatformIO.

closed 2024-02-21 by valeros