platformio/platform-espressif32 #1313: platformio File System erased when upload hex

danielvegafer, 2024-02-19:
Hello. Im developing in platformio for a while right now. My current application is about WiFi server. Everything works great. But… I seems that every time i upload my elf file with upload option, my wifi web page files in spiff partition are also erased. There are some way to configure upload options so platformio dont erases all the flash or at least avoid to erase spiffs partition when upload my program to esp32?
My code is mostly similar to internet examples of AsyncServer wifi server page from internet. That is my platform.ini file
[env:esp-wrover-kit]

platform = espressif32

board = esp-wrover-kit

framework = arduino

monitor_filters = default, time, log2file

monitor_speed = 115200

build_flags = -Os

build_unflags = -O2cntp

board_build.filesystem = littlefs

I test both spiffs and littlefs
At some point i test to use a custom partition table with .csv file, due to unrelated issue, but i left because it don’t fulfill my expected targets, i don’t use anymore these csv indication in the ini file, so i don’t believe there are related to these, i just mention to supply all the information possible.

Jason2866, 2024-02-19:
Do you have a link to the project? Using LittleFS too, and flash over the firmware with the Pio function flash. The FS and the content isn't erased or touched.

danielvegafer, 2024-02-21:
> Do you have a link to the project? Using LittleFS too, and flash over the firmware with the Pio function flash. The FS and the content isn't erased or touched.
#include "WifiServerController.h"
// Globals
#include "system_global.h"
#include "system_global_types.h"

#include "debugs_controller.h"
#include "Wifiutils.h"
// Search for parameter in HTTP POST request
const char *PARAM_INPUT_1 = "input1";

// Variables to save values from HTML form
// String pilot;
// String pass;
String comando;
String MessageR("igual");
String LastMessage("igual");
String Message1 = "Message1";
String Message2 = "Message2";
String Message3 = "Message3";
String Message = "Message";
SMSData sms;
PhonebookData phone;
// IPAddress localIP;
IPAddress localIP(192, 168, 1, 200); // hardcoded

// Set your Gateway IP address
// IPAddress localGateway;
IPAddress localGateway(192, 168, 1, 1); // hardcoded
IPAddress subnet(255, 255, 255, 0);
String ip;
String gateway;
bool authenticated = false;
unsigned long previousMillis, currentMillis = 0;
const long interval = 10000; // interval to wait for Wi-Fi connection (milliseconds)
String PARAM_MESSAGE = "status";
// AsyncWebServer* WifiServerController::_wifi_server = nullptr;
AsyncWebServer WifiServerController::_wifi_server(80);

WifiServerController::WifiServerController()
{
  //_wifi_server = new AsyncWebServer(WIFI_SERVER_PORT);
}

WifiServerController::~WifiServerController()
{
  // delete _wifi_server;
}
/**
 * @brief retrieve Wifi Server
 *
 */
AsyncWebServer *WifiServerController::getServer()
{
  // return _wifi_server;
}
/**
 * @brief  Initializing Wifi Server
 *
 * @return true Wifi Server Initiated succesfully
 * @return false Wifi Server Initiation Error
 */
bool WifiServerController::init()
{
#ifdef WIFI_DEBUG_DEEP
  PRINTLN_N("**** --> Initializing Wifi server...");
#endif
#ifdef WIFI_DEBUG_TIME
  unsigned long routine_start_time = millis();
#endif
[... 526 lines cut ...]

#ifdef WIFI_DEBUG_DEEP
          PRINTLN_N("Wifi command not identified");
#endif
          return;
        }
      }
      else
      {

#ifdef WIFI_DEBUG_DEEP
        PRINTLN_N("Wifi command not identified");
#endif
        return;
      }
    }
  }
  else
  {
#ifdef WIFI_DEBUG_DEEP
    PRINTLN_N("Wifi command not found");
#endif
    return;
  }
}

danielvegafer, 2024-02-21:
I test both littlefs and spifss. when using littlefs, internal log of littlefs or asyncwebserver library, i dont know for sure, reports not exist from old files who are not any more in data folder and even i not request on AsyncWebserver request anymore in these stage of project, so i dont know why check for exist of this files. Every time i upgrade esp32 with new hex , when i try to open the web site after successful connection, the load of the web stucks in the middle and never occurs, and when i build again file system image and upload the image everything works ok

closed 2025-04-25 by stale[bot]