platformio/platform-espressif32 #1314: Wrong longcalls flag in 'ASFLAGS'

hexeguitar, 2024-02-20:
Encountered this problem while trying to compile *.S assembly files. The process fails with message:
`xtensa-esp32s3-elf-as: unrecognized option '-mlongcalls'`
After digging deeper i found that the `xtensa-esp32s3-elf-as` accepts `-longcalls` and `-no-longcalls` flags only, without the -m used by the cpp compiler.
I did try to remove the `'-mlongflags'` from `'ASFLAGS'` using extra scripts, however with no avail. It seems the `-mlongcalls` flag is added anyway in the build process.

I was also able to confirm that the assembly files do compile correctly when the command is invoked from terminal with correct flag set.

hexeguitar, 2024-02-21:
Oh, you're right. My files had lowercase extension. With uppercase all works fine. 
Interestingly, the lowercase "s" files are also getting picked up by the asm compiler, yet always with the wrong flag. This happens on Linux and Windows.
To test it even an empty `testasm.s` file in the project triggers it, `pio run -v` output:
```
No dependencies
Building in release mode
xtensa-esp32s3-elf-as -mlongcalls -o .pio/build/esp32dev/src/testasm.s.o src/testasm.s
xtensa-esp32s3-elf-as: unrecognized option '-mlongcalls'
```

valeros, 2024-02-22:
The problem here is that IDF's build system used behind the scene doesn't provide default assembly flags as they always run both `.S` and `.s` (assembly with and without preprocessing respectively) through generic compiler. Their approach differs a bit from ours, so I've pushed a small fix to the dev branch, so that the build environment for the `.s` files won't contain any build flags by default. It means that you will need to set raw assembly flags via an extra script run in `pre:` mode, for example:
```ini
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
extra_scripts = 
    pre:add_asmflags.py
```
`add_asmflags.py`:
```python
Import ("env")

env.Append(
    ASFLAGS=["--longcalls"]
)
```