platformio/platform-espressif32 #1318: Board request for Waveshare mini ESP32-S3FH4R2 (ESP32-S3-Zero Development Board)

terrafirma2021, 2024-02-27:
Hi, can you please add support for the waveshare mini.

URL: https://www.waveshare.com/esp32-s3-zero.htm

Thanks

valeros, 2024-02-27:
Hi @terrafirma2021, usually a variant header file with proper pin definitions in the [Arduino core](https://github.com/espressif/arduino-esp32) is required to avoid frustrating issues when default pins (e.g. LED_BUILTIN) doesn't correspond to the real hardware.

coryking, 2024-03-04:
In general, how do you know what sorts of things get added to that json file under ./boards? I purchased what looks to be this same board on aliexpress and I can't find much info about it beyond what was in the graphic on their product page.

[image]

coryking, 2024-03-04:
Could it be the same board as this one: https://www.wemos.cc/en/latest/s3/s3_mini.html ?

It looks different, but has the same S3FH4R2 in it and seems to have the same specs:

```
based ESP32-S3FH4R2
2.4 GHz Wi-Fi
Bluetooth LE
4MB Flash
2MB PSRAM
27x IO
1x RGB LED (IO47)
ADC, DAC, I2C, SPI, UART, USB OTG
```

Though the LED is on GP21

terrafirma2021, 2024-03-06:
Hi, can confirm that waveshare state to use on the IDE 
esp32s3.name=ESP32S3 Dev Module
( this is a generic board from what i understand looking at the board info 
[image]

But platformio does not support this native board.

I did look on the website and it was mentioned to use the

terrafirma2021, 2024-03-27:
@coryking Sorry i never reached back to you sooner, i decided to give the board another test on PIO.
https://www.waveshare.com/esp32-s3-zero.htm
You were spot on, thanks for replying, hopefully staff may add copy the config and add the name of the board, which will help out others who find this 5 usd board online and who want to use it without being limited to location of a pio file. 

Can i suggest a name for the board as:
waveshare_S3_Mini

PIO.ini:
[env:lolin_s3_mini]

platform = espressif32

board = lolin_s3_mini

framework = arduino

monitor_speed = 115200

build_flags = -DARDUINO_USB_CDC_ON_BOOT=1

4piu, 2024-05-16:
Note the built-in WS2812 pin on Waveshare S3 Zero is **21**, while on Lolin it's **47**

dzalf, 2024-05-26:
+1 on this request. I am facing issues with the same chip ESP32-S3FH4R2 on a similar board, the [WeAct Studio ESP32-S3-Mini](https://github.com/WeActStudio/WeActStudio.ESP32S3-MINI/blob/main/Hardware/ESP32_S3_MINI_Sch.pdf) and I would like to know what settings are appropriate. Nothing works with Platformio, only with the Arduino IDE as commented by  @terrafirma2021 

Cheers

dixonbg, 2024-09-12:
+1 for ESP32-S3FH4R2 and Waveshare S3 Zero
there are more boards around with this chip https://www.lilygo.cc/products/t-micro32-s3
[image]
https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf

sivar2311, 2024-09-12:
For the ESP32-S3 FH4R2 please see https://github.com/sivar2311/ESP32-S3-PlatformIO-Flash-and-PSRAM-configurations?tab=readme-ov-file#esp32-s3-fh4r2

Godeffroy, 2024-11-21:
+1 for this one as well : [ESP32-S3-Tiny (ESP32-S3FH4R2)](https://www.waveshare.com/wiki/ESP32-S3-Tiny)
RGB LED is on Pin 38

In PlatformIO i can make it work with this config :

```
[platformio]
name = ESP32-S3-Tiny
description = ESP32-S3-Tiny
default_envs = esp32

[env]
; Flash: 4MB QD, PSRAM: 2MB QD
platform = espressif32
board = dfrobot_firebeetle2_esp32s3
framework = arduino
board_build.arduino.memory_type = qio_qspi
board_build.flash_mode = qio
board_build.psram_type = qio
board_upload.flash_size = 4MB
board_upload.maximum_size = 4194304
board_build.extra_flags = 
	-DBOARD_HAS_PSRAM
	;https://docs.espressif.com/projects/arduino-esp32/en/latest/api/usb_cdc.html
	;DARDUINO_USB_CDC_ON_BOOT=1 : Serial used for USB CDC
	-D ARDUINO_USB_CDC_ON_BOOT=1
	;ARDUINO_USB_MODE=1 : USB is in device mode
	-D ARDUINO_USB_MODE=1
build_type = debug
monitor_speed = 921600
monitor_filters = esp32_exception_decoder

[env:esp32]
upload_protocol = esptool
build_flags = -D NO_WIFI
lib_deps = adafruit/Adafruit NeoPixel@^1.12.3
```

samm928, 2024-11-26:
Did anyone get FastLED library to work with this zero dev board ? .. I can compile but the board keeps re-booting. Legacy driver conflict with ng_driver and other nonsense BS like that..

Queequack, 2025-01-11:
Hi @goldiee36 and hi @samm928,

once I had found the right configuration and the upload was working, it was no longer a problem to integrate FastLED. 
Maybe this will help: [issue #1529](https://github.com/platformio/platform-espressif32/issues/1529) (full code available).

Regards René

goldiee36, 2025-01-11:
Many thanks, actually i also found a working config, i had to realise that the specs from the aliexpress vendor i bought were completely wrong.

samm928, 2025-01-12:
That sounds like great news, I can't wait to try it since I have a bunch of
S3 zero I wanted to use for a project. The C3 unfortunately has a die issue
from Expressif they put out an errata sheet. The A5 input doesn't work as
an ADC2  .. all other inputs are on ADC1.

samm928, 2025-01-12:
I got a reply from a message board, someone put in a ticket for a known
issue.
"once I had found the right configuration and the upload was working, it
was no longer a problem to integrate FastLED.
Maybe this will help: issue #1529
<https://github.com/platformio/platform-espressif32/issues/1529> (full code
available)."

Waw, after a year, they actually fix it ..

I verified the library fix for the Waveshare ESP32 S3 Zero now works with
the latest FastLED ver. 3.9.10
I will order some boards + assembly, I just need to check JLCPCB to see if
all parts are still in STOCK.
Only the 4 headers and 3 SOIC8 ICs I have to solder by hand.

[image: 5dfc4870-9329-4c92-b734-a42334468045.png]

[image: 21_band_Analyzer_ESP32_SMDs_SCH.PNG]

TheMechanos, 2025-02-14:
Hello, i buyed ESP32-S3-Zero board and I have problem. This board can't connect to WiFi network... And I can't serve AP with it. Same program flashed into another ESP32S3 board works fine. I have three pieces and there is the same problem for all. Meybe it's hardware problem. Did you manage to get wifi working on this board?

sivar2311, 2025-02-14:
@TheMechanos [duplicate](https://community.platformio.org/t/esp32-s3-zero-does-not-work-on-platformio/40297/16)

samm928, 2025-02-19:
Yes seems like they fix the board library for Waveshare S3 Zero, also
FastLED v.3.9.11 works now, without re-booting non stop. The C3-Mini A0-A4
use ADC1 which works okay but the A5 in ADC2 is broken inside the bare
metal. Expressif screwed up the controller. I checked the Errata sheet.
Check-out my YT channel 'samm928'

Rgds,
Dan