platformio/platform-espressif32 #1323: Waveshare ESP32-S3-Touch-LCD-1.28 board request

birdman3131, 2024-03-06:
Requesting support to be added for this board.

https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28 is the manufactures documentation for it.

valeros, 2024-03-07:
Hi @birdman3131, usually a variant header file with proper pin definitions in the [Arduino core](https://github.com/espressif/arduino-esp32) is required.

martinroger, 2024-04-26:
[esp32s3roundlcd.json](https://github.com/platformio/platform-espressif32/files/15133834/esp32s3roundlcd.json)
I've gotten somewhere defining this as the custom board in the /boards folder of my project and adding board = esp32s3roundlcd to my platformio.ini file.
It would be nice to see this board added to the list of supported boards, as I think I probably did it wrong on my memory size calculations (am completely lost on the relationship between SRAM/PSRAM/ROM/Flash). So that part is probably wrong in my JSON.

closed 2026-10-16 by valeros