platformio/platform-espressif32 #1325: Add support for DFRobot Firebeetle2 ESP32-C6

brutzler, 2024-03-13:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------
### Configuration

**PlatformIO Version** (`platformio --version`):
Core: 6.1.13
### Description of problem

Would be quite nice if this board could also be added to the Platform IO core.

DanielTerletzkiy, 2024-04-20:
is there any ETA for this?

ireun, 2024-06-05:
https://github.com/espressif/arduino-esp32/releases/tag/3.0.0

Jason2866, 2024-06-05:
https://github.com/espressif/arduino-esp32/releases/tag/3.0.1