platformio/platform-espressif32 #1326: Request to support ESP IDF v5.2.1

ascillato, 2024-03-13:
Hello,

Espressif has released on Mar 6th of 2024 the version v5.2.1 of the SDK for the ESP32 family.

https://github.com/espressif/esp-idf/releases/tag/v5.2.1

How can I try this new SDK using PlatformIO?

I tried in _platformio.ini_ unsuccesfully the following:
```ini
platform_packages                       = framework-espidf @ ~3.50201
```

Is there a way that I can generate that package locally?

I'm interested on trying this new SDK version due to the fact that it has several bugfixes related to ADC such as **_Fixed wrong results after ADC continuous mode stops and starts again_**

Thank you a lot in advance.

Jason2866, 2024-03-14:
Hi @ascillato 
you can use this one, until a package is available from Platformio.
```python
platform_packages = https://github.com/tasmota/esp-idf/releases/download/v5.2.1/esp-idf-v5.2.1.zip
```

ascillato, 2024-03-14:
Oh great!!! Thanks. That is awesome.

ascillato, 2024-03-14:
This new version of SDK solves an old issue when the WiFi router channel config is in auto and it jumps to another channel. Now with this SDK, it follows that change. Also it adds more WPA3 protocols support.

Just for the bugfix on the auto channel it's worths to upgrading.

Jason2866, 2024-03-15:
@mhaberler Who i am to suggest something ;-) Official support for newer stuff from espressif needs an contract between espressif and Platformio. This is how i understand the statements from #1225
I will not do an fork which is based and uses the orig. espressif releases and sources.
The reason why i have "something" from time to time it is a side effect of my work to maintain the customized Arduino framework for project Tasmota.
My reasons: 
 - **can't spend the time for**
 - don't have the deep knowledge, to do this well
 - don't want to disturb the contract negotiations between platformio and espressif

ascillato, 2024-03-16:
@Jason2866 Just to learn, how did you make https://github.com/tasmota/esp-idf/releases/download/v5.2.1/esp-idf-v5.2.1.zip ? what differs from https://github.com/espressif/esp-idf/releases/download/v5.2.1/esp-idf-v5.2.1.zip ?

ascillato, 2024-03-16:
@valeros 

Hi,

Is not clear to me what was discussed on https://github.com/platformio/platform-espressif32/issues/1225. Are planned new updates of [platform-espressif32](https://github.com/platformio/platform-espressif32) with new esp-idf versions like the one mentioned in this issue?

Jason2866, 2024-03-16:
> Just to learn, how did you make the package https://github.com/tasmota/esp-idf/releases/download/v5.2.1/esp-idf-v5.2.1.zip ?
> 
> What is different from Espressif release https://github.com/espressif/esp-idf/releases/download/v5.2.1/esp-idf-v5.2.1.zip ?
> 
> EDIT: oh, I see [espressif/esp-idf@master...tasmota:esp-idf:master](https://github.com/espressif/esp-idf/compare/master...tasmota:esp-idf:master) [espressif/github-actions@master...Jason2866:github-actions:master](https://github.com/espressif/github-actions/compare/master...Jason2866:github-actions:master)

The change, just add a package.json. The adding of the version.txt file should not be necessary. Had issues when using the Arduino Lib Builder when the file is not there, so i add by default. Zip (tar) all (including recursive clone of the gitmodules). That's it.
I remove the complete git related stuff, so the package is smaller than 100Mb.

mhaberler, 2024-03-16:
> @mhaberler Who i am to suggest something ;-) Official support for newer stuff from espressif needs an contract between espressif and Platformio. 

ok, we can limp along a bit longer.

but now let's assume this contract does not happen at all, or within a realistic timeline.

espressif support going the same way like Zephyr, i.e. not really? 
defect to another IDE? which one? 

I have no idea what the Espressif folk are thinking with their Arduino strategy - it makes sense only when you want to be the ruler of the low-clue, single-sketch "my LED blinks!" space - I dont get it

valeros, 2024-03-16:
Hi there, support for IDF v5.2.1 is planned to be added this month.

commit 62f67a61 (2024-03-21): Update ESP-IDF to v5.2.1

Resolve #1326
closed 2024-03-21 by valeros