platformio/platform-espressif32 #1327: Custom partition table not working on v6.4.0 and v6.5.0

davcmendes, 2024-03-13:
Hi!

I'm using a custom partition table on one of my esp32 projects, to be more specific the [default_8MB.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default_8MB.csv) table.

Since version 6.4.0 I get the following default 4MB flash usage after compilation:

```
RAM:   [          ]   3.2% (used 10604 bytes from 327680 bytes)
Flash: [=         ]  13.3% (used 174349 bytes from 1310720 bytes)
```

Instead of the normal 8MB flash usage I was getting in version 6.3.2:

```
RAM:   [          ]   3.1% (used 10092 bytes from 327680 bytes)
Flash: [=         ]   5.3% (used 178405 bytes from 3342336 bytes)
```
This is the platformio.ini file I'm using:

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32@6.5.0
board = esp32dev
framework = espidf
board_upload.flash_size = 8MB
board_upload.maximum_size = 8388608
board_build.partitions = default_8MB.csv
```

Was there anything related to partition tables that changed in version 6.4.0, that can explain this behaviour?

I'm attaching a default project where I was able to reproduce the issue.
[test_idf_partition.zip](https://github.com/platformio/platform-espressif32/files/14591967/test_idf_partition.zip)

valeros, 2024-03-14:
Hi @davcmendes, you have two app partitions:
```
app0,     app,  ota_0,   0x10000, 0x140000,
app1,     app,  ota_1,   0x340000,0x330000,
```
By default, the first one is used to determine available size. If you need the other one, just specify it's name directly in your` platformio.ini`:
```ini
[env:esp32dev]
platform = espressif32@6.5.0
board = esp32dev
framework = espidf
board_upload.flash_size = 8MB
board_upload.maximum_size = 8388608
board_build.partitions = default_8MB.csv
board_build.app_partition_name = app1
```

closed 2024-03-14 by valeros

davcmendes, 2024-03-18:
Ok, that works. In the older versions It would always show the available size based on the bigger partition, that’s what was confusing me. Thanks for your help @valeros