platformio/platform-espressif32 #1328: how to reboot board use command?

jack-liew, 2024-03-14:
I work on esp32-s3, it's little hard to press the reset and boot button. how to reboot the board use a command?
like: `pio reboot esp32-board`

Thanks for great work.

valeros, 2024-03-14:
Hi @jack-liew, there is no such functionality at the moment, but you can try to declare your own target via an extra script. This target will trigger `esptoolpy` with a specific set of commands:

```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
framework = arduino
board = esp32-s3-devkitc-1
extra_scripts = 
    reset_target.py
```

reset_target.py:
```python
import os

Import("env")

env.Replace(
    RESETTOOL=os.path.join(
        env.PioPlatform().get_package_dir("tool-esptoolpy") or "", "esptool.py"
    ),
    RESETFLAGS=[
        "--no-stub",
        "--chip",
        env.BoardConfig().get("build.mcu", "esp32"),
        "--port",
        "$UPLOAD_PORT",
        "flash_id",
    ],
    RESETCMD='"$PYTHONEXE" "$RESETTOOL" $RESETFLAGS',
)

env.AddCustomTarget(
    name="reset_target",
    dependencies=None,
    actions=[
        env.VerboseAction(env.AutodetectUploadPort, "Looking for target port..."),
        env.VerboseAction("$RESETCMD", "Resetting target"),
    ],
    title="Reset ESP32 target",
    description="This command resets ESP32 target via esptoolpy",
)
```

Then you can run `pio run -t reset_target`

closed 2024-03-14 by valeros

jack-liew, 2024-03-19:
> Then you can run `pio run -t reset_target`

It's worked, wondeful solution.

Thank U Valeros and PIO.