platformio/platform-espressif32 #1337: write_flash error with Development version

FranckFreiburger, 2024-03-21:
Hello,

after changing from _Stable version_ `platform = espressif32` 
to _Development version_ `platform = https://github.com/platformio/platform-espressif32.git`
I encounter the following error:

```
Uploading .pio\build\esp32dev_DEBUG\firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: C:\work\electro\tests\testJerry\.pio\build\esp32dev_DEBUG\partitions.bin
*** [upload] Error 2
```
using the following partition file:
```
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs      , data  , nvs   ,         , 0x3000
otadata  , data  , ota   ,         , 0x2000
phy_init , data  , phy   ,         , 0x1000
ota_0    , app   , ota_0 ,         , 1600K
storage  , data  , fat   , 0x2ff000, 1024K
```

platformio.ini:

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
framework = espidf

board = esp32dev
board_build.flash_mode = qio
board_build.f_flash = 80000000L
upload_speed = 921600

monitor_speed = 115200
monitor_filters = direct, esp32_exception_decoder

board_build.partitions = default.csv
board_build.filesystem = fatfs
board_build.data_dir = data

build_type = release
build_flags = -DRELEASE -DNDEBUG 

```

Thank you in advance for your help.

valeros, 2024-03-21:
Hi @FranckFreiburger, it's strange, but I cannot reproduce the issue. Could you please open a new terminal window in the IDE, run the upload command in verbose mode `pio run -t upload --verbose` and share here the exact esptool command.

FranckFreiburger, 2024-03-21:
just note that I have have 2 configurations in my `platformio.ini` (see above)
the `[env:esp32dev]` uploads properly whereas `[env:esp32dev_DEBUG]` fails with the error

Here is the result of `pio run -t upload --verbose`:

```

.pio\build\esp32dev\firmware.elf  :

section                                                                                 size         addr

.rtc.text                                                                                  0   1074528256

.rtc.dummy                                                                                 0   1073217536

.rtc.force_fast                                                                            4   1073217536

.rtc_noinit                                                                                0   1342177280

.rtc.force_slow                                                                            0   1342177280

.rtc_fast_reserved                                                                         0   1073225728

.rtc_slow_reserved                                                                        24   1342185448

.iram0.vectors                                                                          1027   1074266112

.iram0.text                                                                            92863   1074267140

.dram0.data                                                                            14164   1073414144

.ext_ram_noinit                                                                            0   1065353216

.noinit                                                                                    0   1073428308

.ext_ram.bss                                                                               0   1065353216

.dram0.bss                                                                            154048   1073428312

.flash.appdesc                                                                           256   1061158944

.flash.rodata                                                                         156852   1061159200

.flash.rodata_noload                                                                   15054   1061316052

.flash.text                                                                           895887   1074593824

.iram0.text_end                                                                            1   1074360003

.iram0.data                                                                                0   1074360004

.iram0.bss                                                                                 0   1074360004

.dram0.heap_start                                                                          0   1073582360

.xt.prop                                                                              746232            0

.xt.lit                                                                                17912            0

.xtensa.info                                                                              56            0

.comment                                                                                 285            0

.xt.lit._ZN3nvs4Lock4initEv                                                                0            0

.xt.prop._ZN3nvs4Lock4initEv                                                              72            0

.xt.prop._ZN14intrusive_listI14NVSHandleEntryE5eraseENS1_8iteratorE                       84            0

.xt.lit._ZN14intrusive_listIN3nvs7Storage14NamespaceEntryEE17clearAndFreeNodesEv           0            0
[... 662 lines cut ...]
BeforeUpload(["upload"], [".pio\build\esp32dev_DEBUG\firmware.bin"])
Auto-detected: COM4
"C:\Users\USER\.platformio\penv\Scripts\python.exe" "C:\Users\USER\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM4" --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 C:\...\my_project\.pio\build\esp32dev_DEBUG\bootloader.bin 0x8000 C:\...\my_project\.pio\build\esp32dev_DEBUG\partitions.bin 0xc000 C:\...\my_project\.pio\build\esp32dev_DEBUG\ota_data_initial.bin 0x10000 .pio\build\esp32dev_DEBUG\firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x8000 for file: C:\...\my_project\.pio\build\esp32dev_DEBUG\partitions.bin
*** [upload] Error 2
====================================================== [FAILED] Took 8.04 seconds ======================================================

Environment     Status    Duration
--------------  --------  ------------
esp32dev        SUCCESS   00:00:23.178
esp32dev_DEBUG  FAILED    00:00:08.039
================================================= 1 failed, 1 succeeded in 00:00:31.217 =================================================
C:\...\my_project>pio run -t upload --verbose

```

Jason2866, 2024-03-22:
The partition table looks not common. https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/partition-tables.html

valeros, 2024-03-22:
I have a suspicion that the size of the bootloader image is greater than 0x7000 bytes in debug mode. Are you using the `BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG` option in your sdkconfig?

FranckFreiburger, 2024-03-22:
file `sdkconfig.esp32dev_DEBUG` :
`# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set`

FranckFreiburger, 2024-03-22:
> The partition table looks not common. https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/partition-tables.html

I rely on automatic offset determination (I hope I've understood correctly how it works.)

valeros, 2024-03-22:
Could you please share here the sizes of all binaries in bytes:
```
.pio\build\esp32dev_DEBUG\bootloader.bin
.pio\build\esp32dev_DEBUG\partitions.bin
.pio\build\esp32dev_DEBUG\ota_data_initial.bin
.pio\build\esp32dev_DEBUG\firmware.bin

```

Jason2866, 2024-03-22:
Placing nvs and otadata at this addresses is not common. As already mentioned your bootloader maybe is bigger than the tool is reserving space for and so it conflicts. Place nvs and ota to the addresses mentioned in examples. So your bootloader has more free address space to use.
```
# Name,   Type, SubType,  Offset,   Size,  Flags
nvs      , data  , nvs   ,         , 0x3000
otadata  , data  , ota   ,         , 0x2000
phy_init , data  , phy   ,         , 0x1000
ota_0    , app   , ota_0 ,         , 1600K
storage  , data  , fat   , 0x2ff000, 1024K
```

FranckFreiburger, 2024-03-22:
same issue, in "debug" mode with the following partition:
```
nvs      , data  , nvs   ,  0x9000 , 0x4000
otadata  , data  , ota   ,  0xd000 , 0x2000
phy_init , data  , phy   ,  0xf000 , 0x1000
ota_0    , app   , ota_0 ,         , 1600K
storage  , data  , fat   , 0x2ff000, 1024K
```

.pio\build\esp32dev_DEBUG\bootloader.bin -> 28 704 bytes
.pio\build\esp32dev_DEBUG\partitions.bin -> 3 072 bytes
.pio\build\esp32dev_DEBUG\ota_data_initial.bin -> 8 192 bytes
.pio\build\esp32dev_DEBUG\firmware.bin -> 1 355 712 bytes

Jason2866, 2024-03-22:
bootloader is too big 28704 = 0x7020.  Esp32 bootloader is flashed at 0x1000
So you have reached the border with 0x8020. Partition table is by default at 0x8000.
You have to change the partition table start in sdkconfig settings.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-partition-table-offset

valeros, 2024-03-22:
I'm also curious if `-D DEBUG` enables some additional code in bootloader sources that causes a larger footprint.

FranckFreiburger, 2024-03-22:
ok, in "release" mode bootloader is 25 152 bytes.
I wondering if it is possible to build bootloader in "release" and firmware in "debug" ?

FranckFreiburger, 2024-03-22:
in my platformio.ini:

`;build_flags = -D DEBUG`
bootloader: 26 992 bytes

`build_flags = -D DEBUG`
bootloader: 28 704 octets

valeros, 2024-03-22:
How is the global `-D DEBUG` macro used in your codebase? If it's only needed by your own codebase I would recommend using `build_src_flags` instead of `build_flags`.

FranckFreiburger, 2024-03-23:
Thanks to the advice !
I still don't understand why bootloader is bigger in `build_type` debug than release (despite the use of `build_src_flags` instead of `build_flags`)
I then changed CONFIG_PARTITION_TABLE_OFFSET to 0x9000 and everything flash fine.

closed 2025-04-25 by stale[bot]