platformio/platform-espressif32 #1340: How do I turn on the power management function?

BeauPeng, 2024-03-25:
I create the project based on vscode + platformio + arduino, there is no sdkconfig and Kconfig.projbuild files, so I can not get the config menu with the instruction--pio run -t menuconfig  ,so I want to know how can open the power management function via menuconfig or not.
[image]

BeauPeng, 2024-03-26:
> Most settings are fixed when using Arduino. The answer is you can't do. If you need more control you have to use Arduino as a component of IDF (described in documentation).

Jason Thanks,  as component of IDF(described in documentation), I want to know which documentation and how I can get it. waitting for your feedback

valeros, 2024-03-26:
Official docs here https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html
PlatformIO example for a  mixed IDF/Arduino project is here https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan

closed 2024-03-26 by valeros