platformio/platform-espressif32 #1341: zigbee not working with platformio/framework-espidf@~3.50102.0

luar123, 2024-03-25:
Building a zigbee example project (basically: https://github.com/espressif/esp-idf/tree/master/examples/zigbee/light_sample/HA_on_off_light for esp32h2) with espidf is working but the zigbee stack is not fully functional, the device won't establish a stable connection.

As it is working with vanilla idf.py from espressif/esp-idf 5.1.2, I copied the components and tools folder from espressif/esp-idf into my  `.platformio/packages/framework-espidf` folder overwriting existing files. This solves the issue and leads to the questions what is the difference?

Settings:
```
[env:h2-test]
board = esp32-h2-devkitm-1
framework = espidf
lib_deps =
    ${common.lib_deps}
lib_ldf_mode = off
platform = https://github.com/Jason2866/platform-espressif32/releases/download/2024.03.10/platform-espressif32-2024.03.10.zip
platform_packages =
    platformio/framework-espidf@~3.50102.0
```

valeros, 2024-03-25:
Hi @luar123, this issue tracker is dedicated to the official integration of the ESP32 platform. Your issue refers to ESP32-H2 family which is not officially supported, so I'd recommend moving your question to appropriate repository https://github.com/Jason2866/platform-espressif32

closed 2024-03-25 by valeros

Jason2866, 2024-03-25:
Hi @luar123 you are using a fork from me. So not an official build. Wrong place to open an issue at official Platformio repo. My fork is not fully compatible to espressif nor Platformio.
The non working zigbee is as designed and intended in the fork.

luar123, 2024-03-25:
Thanks for your replies and sorry, I thought that I am using the default platformio/framework-espidf, but probably that is wrong then? Or is the framework modified by the platform afterwards?
Maybe I should try to make the esp32-H2 work with the official integration, because in general it seems to work.

Jason2866, 2024-03-25:
Use the official solution. There is no support planned for zigbee.

luar123, 2024-03-25:
Just in case someone is looking for this: https://github.com/luar123/platform-espressif32H2/releases/tag/v6.5.1