platformio/platform-espressif32 #1342: Add board OLIMEX ESP32-PoE WROVER

sepp89117, 2024-03-26:
Hi, I added the board for Arduino IDE with following changes. Can you also add the board to platformio?
Additions in `boards.txt`:
```
esp32-poe-wrover.name=OLIMEX ESP32-PoE WROVER

esp32-poe-wrover.bootloader.tool=esptool_py
esp32-poe-wrover.bootloader.tool.default=esptool_py

esp32-poe-wrover.upload.tool=esptool_py
esp32-poe-wrover.upload.tool.default=esptool_py
esp32-poe-wrover.upload.tool.network=esp_ota

esp32-poe-wrover.upload.maximum_size=1310720
esp32-poe-wrover.upload.maximum_data_size=327680
esp32-poe-wrover.upload.flags=
esp32-poe-wrover.upload.extra_flags=

esp32-poe-wrover.serial.disableDTR=true
esp32-poe-wrover.serial.disableRTS=true

esp32-poe-wrover.build.tarch=xtensa
esp32-poe-wrover.build.bootloader_addr=0x1000
esp32-poe-wrover.build.target=esp32
esp32-poe-wrover.build.mcu=esp32
esp32-poe-wrover.build.core=esp32
esp32-poe-wrover.build.variant=esp32-poe-wrover
esp32-poe-wrover.build.board=ESP32_POE_WROVER

esp32-poe-wrover.build.f_cpu=240000000L
esp32-poe-wrover.build.flash_mode=dio
esp32-poe-wrover.build.flash_size=4MB
esp32-poe-wrover.build.boot=dio
esp32-poe-wrover.build.partitions=default
esp32-poe-wrover.build.defines=

esp32-poe-wrover.menu.FlashFreq.80=80MHz
esp32-poe-wrover.menu.FlashFreq.80.build.flash_freq=80m
esp32-poe-wrover.menu.FlashFreq.40=40MHz
esp32-poe-wrover.menu.FlashFreq.40.build.flash_freq=40m

esp32-poe-wrover.menu.UploadSpeed.115200=115200
esp32-poe-wrover.menu.UploadSpeed.115200.upload.speed=115200

esp32-poe-wrover.menu.PartitionScheme.default=Default
esp32-poe-wrover.menu.PartitionScheme.default.build.partitions=default
esp32-poe-wrover.menu.PartitionScheme.no_ota=No OTA (Large APP)
esp32-poe-wrover.menu.PartitionScheme.no_ota.build.partitions=no_ota
esp32-poe-wrover.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
esp32-poe-wrover.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (Large APPS with OTA)
esp32-poe-wrover.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
esp32-poe-wrover.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080

esp32-poe-wrover.menu.PSRAM.enabled=Enabled
esp32-poe-wrover.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
esp32-poe-wrover.menu.PSRAM.enabled.build.extra_libs=
esp32-poe-wrover.menu.PSRAM.disabled=Disabled
esp32-poe-wrover.menu.PSRAM.disabled.build.defines=
esp32-poe-wrover.menu.PSRAM.disabled.build.extra_libs=

esp32-poe-wrover.menu.DebugLevel.none=None
esp32-poe-wrover.menu.DebugLevel.none.build.code_debug=0
esp32-poe-wrover.menu.DebugLevel.error=Error
esp32-poe-wrover.menu.DebugLevel.error.build.code_debug=1
esp32-poe-wrover.menu.DebugLevel.warn=Warn
esp32-poe-wrover.menu.DebugLevel.warn.build.code_debug=2
esp32-poe-wrover.menu.DebugLevel.info=Info
esp32-poe-wrover.menu.DebugLevel.info.build.code_debug=3
esp32-poe-wrover.menu.DebugLevel.debug=Debug
esp32-poe-wrover.menu.DebugLevel.debug.build.code_debug=4
esp32-poe-wrover.menu.DebugLevel.verbose=Verbose
esp32-poe-wrover.menu.DebugLevel.verbose.build.code_debug=5

esp32-poe-wrover.menu.EraseFlash.none=Disabled
esp32-poe-wrover.menu.EraseFlash.none.upload.erase_cmd=
esp32-poe-wrover.menu.EraseFlash.all=Enabled
esp32-poe-wrover.menu.EraseFlash.all.upload.erase_cmd=-e
```

Added `pins_arduino.h` to new folder `esp32-poe-wrover`:
```
#ifndef Pins_Arduino_h
#define Pins_Arduino_h

#include <stdint.h>

#define EXTERNAL_NUM_INTERRUPTS 16
#define NUM_DIGITAL_PINS        40
#define NUM_ANALOG_INPUTS       16

#define analogInputToDigitalPin(p)  (((p)<20)?(analogChannelToDigitalPin(p)):-1)
#define digitalPinToInterrupt(p)    (((p)<40)?(p):-1)
#define digitalPinHasPWM(p)         (p < 34)

#define ETH_CLK_MODE ETH_CLOCK_GPIO0_OUT // instead of ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

static const uint8_t KEY_BUILTIN = 34;

static const uint8_t TX = 1;
static const uint8_t RX = 3;

#define TX1 4
#define RX1 36

#define TX2 33	// ext2 pin 5
#define RX2 35	// ext2 pin 3

static const uint8_t SDA = 13;
static const uint8_t SCL = 33; // instead of 16

static const uint8_t SS    = 5;
static const uint8_t MOSI  = 2;
static const uint8_t MISO  = 15;
static const uint8_t SCK   = 14;

#define BOARD_HAS_1BIT_SDMMC

#endif /* Pins_Arduino_h */
```

Thank you

sepp89117, 2024-03-26:
Hi @valeros,
Is there already a rough time frame for when the next release will appear? Otherwise I will probably insert the board locally myself to bridge the gap. 
Thank you

valeros, 2024-03-26:
Unfortunately, there is no ETA, so I'd recommend using it locally.