platformio/platform-espressif32 #1345: Invalid manifest format Unknown keys:

lukedukeus, 2024-04-03:
I get the error: `Invalid manifest format Unknown keys: licence` when trying to use arduino as a component with platformio. Everthing works fine when building with the ESP-IDF extension, but when building using the platformio extension, I get this error. 

I can get around this error by specifying `framework = arduino, espidf`, but then I need to delete the arduino folder in components directory to prevent it from being picked up. My end goal is to have a project that can be built by either extension.

Reproducible example: https://github.com/lukedukeus/invalid-manifest-issue

Jason2866, 2024-04-03:
Your setup is wrong. Look at the examples how it has to be.
https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-blink

lukedukeus, 2024-04-03:
@Jason2866 I am trying to create a project that works with both the platformio extension for vscode and the esp-idf extension for vscode. The esp-idf extension expects the arduino component to be in the lib or components directory, while platformio uses its own copy. In my example, I have arduino as a component, and it works building from esp-idf, but not with platformio. When building with platformio, I get errors about the idf_component.yml being incorrect, and that it does not recognize fields like licence. I am trying to find a workaround for this, so that I can create a project that can be built with either extension

Jason2866, 2024-04-03:
In this case you have an IDF project only, since you provide Arduino yourself in the component folder.

lukedukeus, 2024-04-03:
@Jason2866  Is there a way to tell platformio to use the arduino componet I have provided, or is there some way I can tell pio to ignore the folder? I have tried using src_filter and lib_ignore, but they don't seem to work, the arduino component is picked up as long as it has a component.yml and CMakeLists.txt. I can delete them and it works for platformio, but this breaks it for esp-idf.

Jason2866, 2024-04-03:
This example works with Platformio. Using Arduino as component in folder.
https://github.com/Jason2866/LILYGO-T5-4.7-E-Paper_Weather-Station

lukedukeus, 2024-04-03:
Thanks, that works. Unfortunatly it only seems to work with arduino core 2.0.0+tasmota and esp-idf v4.3.0 ( via espressif32 @ 3.3.2). Ideally, I would like to use  espressif32 @ 6.5.0 and idf v5.1+ and arduino core 3.0, as the older version is missing some stuff like mdns. It looks like the only reason I can't is because the copy of IDF that pio is using doesn't like the format of the component.yml on certain components. I can remove the licence field and a few others, and it all builds fine, but I would need to do this every time I cloned the repo, as pio's copy of the idf is stored at user/.platformio. Full error:
```
CMake Error at C:/Users/Luke/.platformio/packages/framework-espidf@3.50102.240122/tools/cmake/component.cmake:250 (message):
  ERROR: 2 problems were found in the manifest file
  C:\Users\Luke\Documents\Development\invalid-manifest-issue\managed_components\espressif__esp-zboss-lib\idf_component.yml:

  Invalid manifest format

  Unknown keys: repository_info

  SUGGESTION: This component may be using a newer version of the component
  manager.

  You can try to update the component manager by running:

      C:\Users\Luke\.platformio\penv\.espidf-5.1.2\Scripts\python.exe -m pip install --upgrade idf-component-manager

Call Stack (most recent call first):
  C:/Users/Luke/.platformio/packages/framework-espidf@3.50102.240122/tools/cmake/build.cmake:574 (__component_get_requirements)
  C:/Users/Luke/.platformio/packages/framework-espidf@3.50102.240122/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:3 (project)
```

I get this error even after running `C:\Users\Luke\.platformio\penv\.espidf-5.1.2\Scripts\python.exe -m pip install --upgrade idf-component-manager` as the error suggests. Can this repo be updated to support the newer component.yml schema?

Jason2866, 2024-04-03:
Interesting error message. Have you used `platform = espressif32 @ 6.6.0`?
Since the latest release uses actual idf-component-manager v1.5.2

lukedukeus, 2024-04-03:
Yes, however then I get the error that arduino only supports version 5.1:

```
CMake Error at components/arduino/CMakeLists.txt:19 (message):

    Arduino-esp32 can be used with ESP-IDF versions between 5.1.0 and 5.1.99,
    but a newer version is detected: 5.2.1.

  Call Stack (most recent call first):

    C:/Users/Luke/.platformio/packages/framework-espidf/tools/cmake/scripts/component_get_requirements.cmake:107 (include)
    C:/Users/Luke/.platformio/packages/framework-espidf/tools/cmake/scripts/component_get_requirements.cmake:127 (__component_get_requirements)        

Call Stack (most recent call first):
  C:/Users/Luke/.platformio/packages/framework-espidf/tools/cmake/build.cmake:578 (__component_get_requirements)
  C:/Users/Luke/.platformio/packages/framework-espidf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:3 (project)
```

lukedukeus, 2024-04-03:
Also, with this `platformio.ini`:
```
[platformio]
src_dir = main
; lib_dir = componenets

[env:esp32dev]
platform = espressif32 @ 6.5.0
platform_packages = framework-espidf @ 3.50102
board = esp32dev
framework = espidf
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
monitor_raw = yes
```

were back to this issue:
```
CMake Error at C:/Users/Luke/.platformio/packages/framework-espidf@3.50102.240122/tools/cmake/component.cmake:250 (message):
  ERROR: 2 problems were found in the manifest file
  C:\Users\Luke\Documents\Development\invalid-manifest-issue\managed_components\espressif__esp-zboss-lib\idf_component.yml:

  Invalid manifest format

  Unknown keys: repository_info
```

Jason2866, 2024-04-03:
Yes, as I wrote in my previous message you have to install IDF 5.1 via `platform_packages`

Jason2866, 2024-04-03:
Use `platform = espressif32 @ 6.6.0`

lukedukeus, 2024-04-03:
Sorry, should have clarified, I have tried both ways. with this `platformio.ini`: 
```
[platformio]
src_dir = main
; lib_dir = componenets

[env:esp32dev]
platform = espressif32 @ 6.6.0
platform_packages = framework-espidf @ 3.50102
board = esp32dev
framework = espidf
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
monitor_raw = yes
```

I get:
```
Reading CMake configuration...

*** Two environments with different actions were specified for the same target: C:\Users\Luke\Documents\Development\invalid-manifest-issue\.pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\support\cplx_gen\dsps_cplx_gen.o
(action 1: xtensa-esp32-elf-gcc -DPLATFORMIO=60114 -DARDUINO_ESP32_DEV -DESP_PLATFORM -DIDF_VER=\"5.1.2\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -c -o dsps_cplx_gen.o dsps_cplx_gen.S)
(action 2: xtensa-esp32-elf-gcc -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=C:/Users/Luke/Documents/Development/invalid-manifest-issue=. -fmacro-prefix-map=C:/Users/Luke/.platformio/packages/framework-espidf@3.50102.240122=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DPLATFORMIO=60114 -DARDUINO_ESP32_DEV -DESP_PLATFORM -DIDF_VER=\"5.1.2\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -c -o dsps_cplx_gen.o dsps_cplx_gen.S)
File "C:\Users\Luke\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 698, in compile_source_files
============================================================================== [FAILED] Took 1.54 seconds ==============================================================================

```

Jason2866, 2024-04-03:
Ahh, progress. Add to the [env]
```
board_build.esp-idf.preserve_source_file_extension = yes
```

Jason2866, 2024-04-03:
If compile starts and you run later in compile errors, probably you have to add to the [env]
```
board_build.embed_txtfiles =
    managed_components/espressif__esp_insights/server_certs/https_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_mqtt_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_claim_service_server.crt
    managed_components/espressif__esp_rainmaker/server_certs/rmaker_ota_server.crt
```

lukedukeus, 2024-04-03:
It gets much further, heres the new error I get at the last line:
```
Processing esp32dev (platform: espressif32 @ 6.6.0; board: esp32dev; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.6.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50102.240122 (5.1.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1 
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
Generating assembly for certificate bundle...
Warning! Flash memory size mismatch detected. Expected 4MB, found 2MB!
Please select a proper value in your `sdkconfig.defaults` or via the `menuconfig` target!
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\main\main.c.o
Generating LD script .pio\build\esp32dev\memory.ld
Compiling .pio\build\esp32dev\components\SomeComponent\SomeComponent.c.o
Compiling .pio\build\esp32dev\app_trace\app_trace.c.o
Compiling .pio\build\esp32dev\app_trace\app_trace_util.c.o
Compiling .pio\build\esp32dev\app_trace\host_file_io.c.o
Compiling .pio\build\esp32dev\app_trace\port\port_uart.c.o
Compiling .pio\build\esp32dev\app_update\esp_ota_ops.c.o
Archiving .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a
Archiving .pio\build\esp32dev\esp-idf\SomeComponent\libSomeComponent.a
Compiling .pio\build\esp32dev\app_update\esp_ota_app_desc.c.o
Indexing .pio\build\esp32dev\esp-idf\app_trace\libapp_trace.a
Indexing .pio\build\esp32dev\esp-idf\SomeComponent\libSomeComponent.a
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\base64.cpp.o
Archiving .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\cbuf.cpp.o
Indexing .pio\build\esp32dev\esp-idf\app_update\libapp_update.a
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\chip-debug-report.cpp.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-adc.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-bt.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-cpu.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-dac.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-gpio.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-i2c.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-i2c-slave.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-ledc.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-matrix.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-misc.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-periman.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-psram.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-rgb-led.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-sigmadelta.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-spi.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-time.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-timer.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-tinyusb.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-touch.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-uart.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\esp32-hal-rmt.c.o
Compiling .pio\build\esp32dev\components\arduino\cores\esp32\Esp.cpp.o
[... 642 lines cut ...]
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\float\dsps_fird_init_f32.c.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\fixed\dsps_fird_init_s16.c.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\fixed\dsps_fird_s16_ansi.c.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\fixed\dsps_fird_s16_ae32.S.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\fixed\dsps_fir_s16_m_ae32.S.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\fir\fixed\dsps_fird_s16_aes3.S.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\kalman\ekf\common\ekf.cpp.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp-dsp\modules\kalman\ekf_imu13states\ekf_imu13states.cpp.o
[... cut ...]
Compiling .pio\build\esp32dev\managed_components\espressif__esp_insights\src\esp_insights_encoder.c.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp_insights\src\esp_insights_cbor_decoder.c.o
Compiling .pio\build\esp32dev\managed_components\espressif__esp_insights\src\esp_insights_cbor_encoder.c.o
*** [.pio\build\esp32dev\.pio\build\esp32dev\https_server.crt.S.o] Source `.pio\build\esp32dev\https_server.crt.S' not found, needed by target `.pio\build\esp32dev\.pio\build\esp32dev\https_server.crt.S.o'.
```

Jason2866, 2024-04-03:
add to your sdkconfig.defaults 
```
CONFIG_AUTOSTART_ARDUINO=y
# CONFIG_WS2812_LED_ENABLE is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_HEADER_FLASHSIZE_UPDATE=y
```

lukedukeus, 2024-04-03:
> If compile starts and you run later in compile errors, probably you have to add to the [env]
>
> ```
> board_build.embed_txtfiles = [...]

Just saw this, yep it works now! Thank you so much!

lukedukeus, 2024-04-03:
> add to your sdkconfig.defaults
>
> ```
> CONFIG_AUTOSTART_ARDUINO=y
> # CONFIG_WS2812_LED_ENABLE is not set
> CONFIG_FREERTOS_HZ=1000
> CONFIG_MBEDTLS_PSK_MODES=y
> CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y [...]

Didn't even need to set these only needed `CONFIG_FREERTOS_HZ=1000`

Jason2866, 2024-04-03:
I would add ;-) Without WiFi will not work.

closed 2024-04-03 by lukedukeus