platformio/platform-espressif32 #1350: Feature request for support for  ESP32 C3 SuperMini

Luriel627, 2024-04-08:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info
Feature request for support for  ESP32 C3 SuperMini
https://banggood.com/ESP32-C3-Development-Board-ESP32-SuperMini-WiFi-Bluetooth-Mini-Module-p-1997449.html

lboue, 2024-04-14:
@Luriel627 
You can add a file in [boards](https://github.com/platformio/platform-espressif32/tree/develop/boards) dir and submit a PR.

RayPS, 2025-06-12:
Before someone create a PR you can try [pioarduino](https://github.com/pioarduino/pioarduino-vscode-ide)
[image]

zaniar, 2026-02-01:
> Before someone create a PR you can try [pioarduino](https://github.com/pioarduino/pioarduino-vscode-ide) <img alt="Image" width="837" src="https://private-user-images.githubusercontent.com/1941540/454164922-aa5b8740-87dc-448b-8c13-1106a6e3b91b.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Njk5MjgzOTksIm5iZiI6MTc2OTkyODA5OSwicGF0aCI6Ii8xOTQxNTQwLzQ1NDE2NDkyMi1hYTViODc0MC04N2RjLTQ0OGItOGMxMy0xMTA2YTZlM2I5MWIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI2MDIwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNjAyMDFUMDY0MTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjM0OTE2NmRjZTUyYmQwZDJiMTAyM2FjMTE2ODQzODE5MzcyNzFmNjk4ZjQwZDZmMjFmZTkyMTRmMGYwNDI0NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.cZduKBSa5-WeCfvdKlGDAxX7EcGndXd3Dsg75I6D3cw">

I tried pioarduino but I can't find **Nologo ESP32C3 SuperMini**.