platformio/platform-espressif32 #1353: gdb version too old - needs python2.7

anm, 2024-04-11:
I can't run esp32 gdb because python2.7 is too old for my OS.
Installing the latest version direct from espressif works.

I've seen other people encountering this problem on platformio forums too.

Jason2866, 2024-04-14:
I have nowhere an issue not using python v2.7 with GDB

calint, 2024-05-10:
Same problem in Ubuntu 24.04.

arduhe, 2024-05-29:
I would really appreciate if this could be fixed soon.  This ancient version of python2 - support for Python 2 ended Jan. 1, 2020 [see announcement on python.org]( https://www.python.org/doc/sunset-python-2/) - may be a security risk.

Jason2866, 2024-05-29:
What is the exact issue? What command is not working without Python2?

arduhe, 2024-05-29:
```
ldd xtensa-esp32-elf-gdb
        linux-vdso.so.1 (0x00007ffc70f64000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f4d1d2f2000)
        **libpython2.7.so.1.0** => /usr/lib64/libpython2.7.so.1.0 (0x00007f4d1d0a7000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f4d1d083000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00007f4d1d07f000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f4d1cf33000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f4d1ccec000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f4d1ccc8000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f4d1cad1000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f4d1d339000)

```

gdb is linked dynamically against V2 of libpython. Several distros do no longer offer V2 - so you have either to install it manuallly - as I did longer ago (and forgot that fact until today 😉) - or you get an error message due to the missing lib.

I saw in the meantime that @valeros has upgraded the python version in CI to 3.9 - so I am little bit optimistic 
that the next release fixes this isuue

Jason2866, 2024-05-29:
Ahh, you are using the old version

arduhe, 2024-05-29:
Not voluntary 😆

arduhe, 2024-06-17:
It's even worse - xtensa-esp32s3-elf-gdb shows unsurprisingly the same behaviour

 ```
ldd ~/.platformio/packages/toolchain-xtensa-esp32s3/bin/xtensa-esp32s3-elf-gdb
        linux-vdso.so.1 (0x00007ffefe7e5000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007fe6980f4000)
        libpython2.7.so.1.0 => not found
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007fe6980f1000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00007fe6980ec000)
        libm.so.6 => /lib64/libm.so.6 (0x00007fe698002000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007fe697c00000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007fe697fde000)
        libc.so.6 => /lib64/libc.so.6 (0x00007fe697800000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fe698123000)

```

But if I install the missed library manually (and nothing else from python2.7)  the linkage error disappears but instead a - at a first glance  - strange python error occurs

`
piodebuggdb  -e esp32S3
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
adapter speed: 40000 kHz

Warn : Transport "jtag" was already selected
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (24:58:7C:CD:A6:48)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Could not find platform independent libraries <prefix>
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
ImportError: No module named site

`

This is IMHO due to missing support of python2.7 on my system. 

I was able to reproduce this behavior under 3 different distributions/installations

- openSUSE Leap 15.6 (using python3.6)
- openSUSE Tumbleweed  (using python3.11)
- debian 12 (using python3.11)

arduhe, 2024-07-09:
For those who are waiting like me for a solution of this issue - in the meantime I've found a workaround described [here](https://community.platformio.org/t/debug-aborts-with-python-error/41139/2) - nevertheless it would be REALLY fine, if the support for PYTHON**3** would be pushed forward ...

green-green-avk, 2024-09-14:
Debian Bookworm is in the same club...

mplogas, 2024-10-20:
Will the updated esp32s3 gdb libraries added to platformio in the foreseeable future or is using the files from the esp-idf toolchain the only option for now?

voidless-pawn, 2025-02-16:
Same issue on Arch/Manjaro, got [removed from the repositories](https://archlinux.org/news/removing-python2-from-the-repositories/) in 2022.
+1 for an update.

YutharsanS, 2025-04-24:
I was able to get debugging working after installing the `libpython3-all-dev` and `libpython3-all-dbg` packages in Ubuntu 24.04.

anm, 2025-06-26:
Is it fixed?