platformio/platform-espressif32 #1355: mbedTLS no spiram

NaxGH, 2024-04-13:
How do I allocate mbedTLS to external memory (PSRAM)? 
In IDF framework you can set mbedTLS allocate to External Memory in menuconfig
It is possible to allocate mbedTLS to spiram in Arduino framework?

Jason2866, 2024-04-14:
Wrong place to ask general questions. This github is for issues with Platformio itself.
You can ask for help here https://community.platformio.org/tag/espressif32

valeros, 2024-04-15:
Hi @NaxGH, the underlying IDF settings in the Arduino framework are static and cannot be changed. As a workaround you can try to use Arduino as a component, here is an [example](https://github.com/platformio/platform-espressif32/tree/develop/examples/espidf-arduino-wifiscan).

closed 2024-04-15 by valeros