platformio/platform-espressif32 #1359: How to run idf.py or coredump.py from Platformio cli?

hitecSmartHome, 2024-04-18:
### Environment

- PLATFORM: Espressif 32 (6.6.0) > Espressif ESP-WROVER-KIT
- HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
- DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
- framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
- framework-espidf @ 3.40406.240122 (4.4.6)
- tool-cmake @ 3.16.4
- tool-esptoolpy @ 1.40501.0 (4.5.1)
- tool-idf @ 1.0.1
- tool-mconf @ 1.4060000.20190628 (406.0.0)
- tool-ninja @ 1.9.0
- tool-openocd-esp32 @ 2.1100.20220706 (11.0)
- toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
- toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5

I want to decode my downloaded coredump bin file.
I have installed ESP-IDF 5.2 PowerShell with all the tools for ESP IDF.
I have cd-d into my project directory which uses Arduino as a component of IDF.

**When i try these commands:**
1. `idf.py coredump-info -c your-coredump.txt`
2. `espcoredump.py info_corefile -c your-coredump.txt -t raw <your_app_elf_file_path>`

They do not work. I assume it is because of the PIO IDF installation. The separate IDF installation is not happy with it.

Here is the relevant issue ticket on ESP IDF. They sent me there. 
#https://github.com/espressif/esp-idf/issues/13611

valeros, 2024-04-18:
Hi @hitecSmartHome, I'm not sure if we can help you in any way because the issues you're describing are related to third-party tools which are not directly supported by PlatformIO. Just curious, since you've already installed IDF development environment, have you tried to compile your project there? Does it work as intended?

hitecSmartHome, 2024-04-18:
I did not tried to compile. The only reason that I installed the IDF is that i wanted to decode my coredump bin file.
I have no intention to use it for anything else. PIO inspect is broken so I might use the IDF code checker too. Since I use PIO's idf and Arduino as a component, is there any way to call it from pio?

valeros, 2024-04-18:
> Since I use PIO's idf and Arduino as a component, is there any way to call it from pio?

The tools you mentioned are not part of the PlatformIO integration, but it doesn't mean it shouldn't work. This behavior may be hiding an issue in our integration, so I'd kindly recommend testing the same project but within the IDF environment, so that we can figure out which part is causing this issue.

closed 2024-08-07 by hitecSmartHome