platformio/platform-espressif32 #1360: Crash when enabling PSRAM moving from 6.5.0 to 6.6.0

western-hoolock, 2024-04-18:
Module: ESP32-WROVER-E (ESP32-D0WD-V3)

My project is literally this:
```
void app_main() {
    while (1) {
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}
```

Works fine with 6.5.0 when external RAM option is enabled in menuconfig (no other defaults changed).
When compiling with 6.6.0 it crashes and produces the following stacktrace:

```
I (1199) heap_init: Initializing. RAM available for dynamic allocation:
I (1205) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1211) heap_init: At 3FFB2A30 len 0002D5D0 (181 KiB): DRAM
I (1218) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1224) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1231) heap_init: At 4008F30C len 00010CF4 (67 KiB): IRAM
I (1237) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x400e72bc  PS      : 0x00060734  A0      : 0x8008c164  A1      : 0x3ffe3a50
A2      : 0x3ff42000  A3      : 0x3ffe3ab0  A4      : 0x00000003  A5      : 0x00000003
A6      : 0x00000000  A7      : 0x00000003  A8      : 0x8008be3d  A9      : 0x3ffe3a20
A10     : 0x3ffe3ab0  A11     : 0x3ffe3a50  A12     : 0x00000003  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffe  SAR     : 0x00000020  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x400e7476  LEND    : 0x400e7481  LCOUNT  : 0x00000000

Backtrace: 0x400e72b9:0x3ffe3a50 0x4008c161:0x3ffe3a80 0x4008de59:0x3ffe3ab0 0x400831e7:0x3ffe3af0 0x400d3571:0x3ffe3b20 0x400832c2:0x3ffe3b40 0x400d33a3:0x3ffe3b70 0x400d18bc:0x3ffe3bd0 0x400d1b0d:0x3ffe3bf0 0x4008155e:0x3ffe3c30 0x400796d1:0x3ffe3c80 |<-CORRUPTED

  #0  0x400e72b9:0x3ffe3a50 in __memcpy_aux at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:130
  #1  0x4008c161:0x3ffe3a80 in spi_flash_hal_common_command at C:\Users\WH\.platformio\packages\framework-espidf\components\hal/spi_flash_hal_common.inc:180
  #2  0x4008de59:0x3ffe3ab0 in memspi_host_read_id_hs at C:\Users\WH\.platformio\packages\framework-espidf\components\spi_flash/memspi_host_driver.c:99
  #3  0x400831e7:0x3ffe3af0 in read_id_core at C:\Users\WH\.platformio\packages\framework-espidf\components\spi_flash/esp_flash_api.c:424
  #4  0x400d3571:0x3ffe3b20 in esp_flash_read_chip_id at C:\Users\WH\.platformio\packages\framework-espidf\components\spi_flash/esp_flash_api.c:442
  #5  0x400832c2:0x3ffe3b40 in esp_flash_init_main at C:\Users\WH\.platformio\packages\framework-espidf\components\spi_flash/esp_flash_api.c:351 (discriminator 2)
  #6  0x400d33a3:0x3ffe3b70 in esp_flash_init_default_chip at C:\Users\WH\.platformio\packages\framework-espidf\components\spi_flash/esp_flash_spi_init.c:367
  #7  0x400d18bc:0x3ffe3bd0 in do_core_init at C:/Users/WH/.platformio/packages/framework-espidf/components/esp_system/startup.c:333
  #8  0x400d1b0d:0x3ffe3bf0 in start_cpu0_default at C:/Users/WH/.platformio/packages/framework-espidf/components/esp_system/startup.c:467 (discriminator 3)
  #9  0x4008155e:0x3ffe3c30 in call_start_cpu0 at C:/Users/WH/.platformio/packages/framework-espidf/components/esp_system/port/cpu_start.c:784
  #10 0x400796d1:0x3ffe3c80 in ?? ??:0
```

Electry, 2024-04-18:
Following as I have the same issue with my esp32-cam board (which is also ESP32-D0WD-V3).
Compiled with 6.5.0 (IDF 5.1) & SPIRAM enabled everything works perfectly fine, 6.6.0 (IDF 5.2) gives me the same exact crash.

Slider0007, 2024-04-19:
I'm also facing this issue using ESP32-D0WD-V0 hardware (ESP32-CAM board).

Note: Comipiling same project with ESP32S3 hardware (XIAO ESP32S3 Sense) -> No issues

valeros, 2024-04-23:
Thanks for reporting, could someone experiencing this issue try the upstream version of the platform?

```ini
[env:esp32dev]
platform = espressif32
framework = espidf
board = esp32dev
```

Slider0007, 2024-04-24:
@valeros:

I've just tested latest development version. ESP32 device is not crashing anymore. Issue seems to be fixed. Thanks for looking into it.

Electry, 2024-04-26:
I can also confirm that the issue has been fixed for my board, using the devel branch with toolchain 13.2.0. Thanks.

western-hoolock, 2024-05-17:
Just tried the latest 6.7.0 release and can confirm it's not crashing anymore when external RAM is configured and used.
Thanks!

closed 2024-05-17 by western-hoolock