platformio/platform-espressif32 #1363: Feature request for ESP32-2432S024C boards.

BrazenCodebane, 2024-04-24:
ISSUE: Implementing Meshtastic on ESP32-2432S024C TFT (commonly known as CYD[Cheap-Yellow-Display]) which uses platformIO. 

board currently unsupported creating a roadblock to the project.

Amazon link to the board:
https://www.amazon.com/gp/product/B0CLGD2DG6/ref=ox_sc_act_title_9?smid=A2RJ79XBQX6W3M&psc=1

valeros, 2024-04-24:
Hi @BrazenCodebane, usually a variant header file with proper pin definitions in the [Arduino core](https://github.com/espressif/arduino-esp32) is required.

Jason2866, 2024-04-24:
> board currently unsupported creating a roadblock to the project.

No, just add a custom boards manifest to the project with a folder name `boards`