platformio/platform-espressif32 #1364: adafruit_qtpy_esp32s3_nopsram board unknown

ChuckVanzant, 2024-04-25:
The board adafruit_qtpy_esp32s3_nopsram is listed as a supported board but when saving the platformio.ini file, the following error message appears:

**Resolving adafruit_qtpy_esp32s3_nopsram dependencies...
UnknownBoard: Unknown board ID 'adafruit_qtpy_esp32s3_nopsram'**

This is what I have specified in my platformio.ini:

**[env:adafruit_qtpy_esp32s3_nopsram]
platform = espressif32
board = adafruit_qtpy_esp32s3_nopsram
framework = arduino**

[image]

[image]

[image]

valeros, 2024-04-25:
Hi @ChuckVanzant, what version of the `espressif32` development platform are you using? This board was added in v6.5.0, so please run `pio pkg update -g -p espressif32` directly in the IDE terminal.

ChuckVanzant, 2024-04-25:
Updated and the board is now recognized. Thanks!

I was just confused as to why the board could have been listed through the Board Explorer and directed to

Adafruit QT Py ESP32-S3 No PSRAM — PlatformIO v6.1 documentation<https://docs.platformio.org/en/latest/boards/espressif32/adafruit_qtpy_esp32s3_nopsram.html?utm_source=platformio&utm_medium=piohome>

when the development platform in use was not correct.

Thanks again!

closed 2024-04-25 by valeros