platformio/platform-espressif32 #1374: How to confirm a pio plaform to use IDF 4.4.7 ?

Koxx3, 2024-05-08:
Hello,
I tried many platformio.ini configurations to use IDF 4.4.7, but nothing works.
Do you plan to release such platform ?
Can anybody help to target this version please ?
Thanks !

Koxx3, 2024-05-08:
thanks a lot !

I am trying to configure a project with IDF 4.4.7 + Arduino 2.0.13 (not above since WifiClientSecure has been removed and I need it), but I have some trouble :

```
platform = espressif32 @ ^5.4.0
platform_packages = 
	framework-espidf @ https://github.com/Jason2866/esp-idf/releases/download/v4.4.7/esp-idf-v4.4.7.zip
	framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.13/esp32-2.0.13.zip

framework = arduino, espidf 
```

IDF 4.4.7 download & install seems to work fine, but after that : 

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (5.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 2.0.13 
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Using ESP-IDF: /home/koxx/.platformio/packages/framework-espidf
Using toolchain: /home/koxx/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5
test
venv_dir /home/koxx/.platformio/penv/.espidf-4.4.7
venv_data_file /home/koxx/.platformio/penv/.espidf-4.4.7/pio-idf-venv.json
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
IndexError: list index out of range:
  File "/home/koxx/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/koxx/.platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/koxx/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/koxx/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Util/envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "/home/koxx/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 676:
    return method(*args, **kw)
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/koxx/.platformio/packages/tool-scons/scons-local-4.7.0/SCons/Script/SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/koxx/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 1523:
    arduino_config_name = list(
```

Jason2866, 2024-05-08:
Just had a look, Platformio has updated the IDF 4.4.x version to 4.4.7. Did miss that
So everything is available from Platformio. Core 2.0.16 and IDF 4.4.7
This setup should work.
```
platform = espressif32 @ ^5.4.0
platform_packages = 
	platformio/framework-espidf @ 4.4.7
	platformio/framework-arduinoespressif32 @ 2.0.16
```

Koxx3, 2024-05-08:
FYI, I found this working configuration, but it seems "dirty" (I had to disable Ethernet component in sdkconfig).
WifiClientSecure was removed only from Tasmota Arduino...

```
platform = espressif32 @ ^5.4.0
platform_packages = 
	tool-mkspiffs @ 2.230.0
	framework-espidf @ https://github.com/Jason2866/esp-idf/releases/download/v4.4.7/esp-idf-v4.4.7.zip
	framework-arduinoespressif32 @ ~3.20016.0
	; framework-arduinoespressif32 @ ~3.20009.0 ; working too
	; framework-arduinoespressif32 @ 2.0.6 ; working too

framework = arduino, espidf
```

Koxx3, 2024-05-08:
> Just had a look, Platformio has updated the IDF 4.4.x version to 4.4.7. Did miss that So everything is available from Platformio. Core 2.0.16 and IDF 4.4.7 This setup should work.
> 
> ```
> platform = espressif32 @ ^6.6.0
> platform_packages = 
> 	platformio/framework-espidf @ ~3.40407.0
> 	platformio/framework-arduinoespressif32 @ ~3.20016.0
> ```
> 
> WifiClientSecure is in every official espressif / Platformio version!

ahahah, excellent. thanks for your feedback.
your setup is much cleaner than mine ;)