platformio/platform-espressif32 #1379: Version 6.7.0 crashes on ESP32 when trynig to init NVS

ascillato, 2024-05-14:
The latest version of **platform-espressif32** [v6.7.0](https://github.com/platformio/platform-espressif32/releases/tag/v6.7.0) makes that any firmware on ESP32 crashes as soon as it tries to init NVS.

**_This bug is not present in v6.6.0_**
Reverting back to 6.6.0 solves this issue and the firmware works as expected.

The firmware tested is:

```cpp
#include "esp_log.h"

extern "C" void app_main() {
    static const char* TAG = "INIT";

    esp_err_t ret = nvs_flash_init();

    ESP_LOGI(TAG, "NVS: %d", ret);
}
```

Error output:

```console
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x4019ee8c  PS      : 0x00060d35  A0      : 0x80082a43  A1      : 0x3ffbd330  
A2      : 0x3ffbe028  A3      : 0xffffffff  A4      : 0x00060d23  A5      : 0x00060d23  
A6      : 0x00000001  A7      : 0x0000cdcd  A8      : 0x800943ed  A9      : 0x3ffbd320  
A10     : 0x3ffbe028  A11     : 0xb33fffff  A12     : 0x0000cdcd  A13     : 0x3ffbd230  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000010  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4019ee89:0x3ffbd330 0x40082a40:0x3ffbd350 0x400901eb:0x3ffbd370 0x400911b6:0x3ffbd390 0x4009002f:0x3ffbd3b0 0x40090109:0x3ffbd3d0 0x40188459:0x3ffbd400 0x4011dbb9:0x3ffbd440 0x401006d5:0x3ffbd480 0x401008a0:0x3ffbd520 0x401008d9:0x3ffbd540 0x4011d700:0x3ffbd560 0x4011c298:0x3ffbd580 0x4011ab94:0x3ffbd5b0 0x4011aba6:0x3ffbd5d0 0x400dbd8b:0x3ffbd5f0 0x400dbdbe:0x3ffbd610 0x400db2e3:0x3ffbd630 0x401a475f:0x3ffbd650
  #0  0x4019ee89 in xt_utils_set_breakpoint at /home/ascillato/.platformio/packages/framework-espidf/components/xtensa/include/xt_utils.h:122
      (inlined by) esp_cpu_set_breakpoint at /home/ascillato/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:339
  #1  0x40082a40 in xPortEnterCriticalTimeoutSafe at /home/ascillato/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:583
      (inlined by) vPortEnterCriticalSafe at /home/ascillato/.platformio/packages/framework-espidf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:592
      (inlined by) esp_ipc_isr_stall_other_cpu at /home/ascillato/.platformio/packages/framework-espidf/components/esp_system/port/esp_ipc_isr.c:113
  #2  0x400901eb in Cache_Flush at /home/ascillato/.platformio/packages/framework-espidf/components/esp_rom/include/esp32/rom/cache.h:140
  #3  0x400911b6 in cache_sync at /home/ascillato/.platformio/packages/framework-espidf/components/esp_mm/cache_esp32.c:29
  #4  0x4009002f in s_do_cache_invalidate at /home/ascillato/.platformio/packages/framework-espidf/components/esp_mm/esp_mmu_map.c:378
  #5  0x40090109 in s_do_mapping at /home/ascillato/.platformio/packages/framework-espidf/components/esp_mm/esp_mmu_map.c:432
  #6  0x40188459 in esp_mmu_map at /home/ascillato/.platformio/packages/framework-espidf/components/esp_mm/esp_mmu_map.c:571
  #7  0x4011dbb9 in spi_flash_mmap at /home/ascillato/.platformio/packages/framework-espidf/components/spi_flash/flash_mmap.c:85
  #8  0x401006d5 in load_partitions at /home/ascillato/.platformio/packages/framework-espidf/components/esp_partition/partition.c:103
      (inlined by) ensure_partitions_loaded at /home/ascillato/.platformio/packages/framework-espidf/components/esp_partition/partition.c:259
  #9  0x401008a0 in ensure_partitions_loaded at /home/ascillato/.platformio/packages/framework-espidf/components/esp_partition/partition.c:306
      (inlined by) esp_partition_find at /home/ascillato/.platformio/packages/framework-espidf/components/esp_partition/partition.c:288
  #10 0x401008d9 in esp_partition_find_first at /home/ascillato/.platformio/packages/framework-espidf/components/esp_partition/partition.c:344
  #11 0x4011d700 in nvs::partition_lookup::lookup_nvs_partition(char const*, nvs::NVSPartition**) at /home/ascillato/.platformio/packages/framework-espidf/components/nvs_flash/src/nvs_partition_lookup.cpp:14
  #12 0x4011c298 in nvs::NVSPartitionManager::init_partition(char const*) at /home/ascillato/.platformio/packages/framework-espidf/components/nvs_flash/src/nvs_partition_manager.cpp:46
  #13 0x4011ab94 in nvs_flash_init_partition at /home/ascillato/.platformio/packages/framework-espidf/components/nvs_flash/src/nvs_api.cpp:135 (discriminator 1)
      (inlined by) nvs_flash_init_partition at /home/ascillato/.platformio/packages/framework-espidf/components/nvs_flash/src/nvs_api.cpp:127 (discriminator 1)
  #14 0x4011aba6 in nvs_flash_init at /home/ascillato/.platformio/packages/framework-espidf/components/nvs_flash/src/nvs_api.cpp:163
```

Please, let me know how I can help on this and if any other information is needed.

ascillato, 2024-05-14:
I think I found what is causing this issue.

One of the changes of 6.7.0 that differs from 6.6.0 is in **_espidf.py_** file:

```cpp
            preserve_source_file_extension = board.get(
                "build.esp-idf.preserve_source_file_extension", True
            )
```

I changed it back to `False` as 6.6.0 has and now it works fine.

Why this change in 6.7.0 is affecting the compiled firmware which leaves the CPU cache disabled?

ascillato, 2024-05-14:
Knowing this, I tried to add to **_platformio.ini_** the line:

```ini
board_build.esp-idf.preserve_source_file_extension = no
```

But seems that it is being overrided by espidf.py and the issue persist. I have to change the espidf.py file in order to make the firmware to work.

@valeros 
Why this change in 6.7.0 is affecting the compiled firmware which leaves the CPU cache disabled?

valeros, 2024-05-15:
Hi @ascillato, thanks for reporting. I've reproduced the issue, please stick to v6.6.0 until I figure out the underlying reason.

ascillato, 2024-05-15:
@valeros 

Hi, thanks for the fast response!

Updating to new 6.7.0 fixes the issue!!!!!!!

Thanks!!!!! Very appreciated.

--------

I had to change io `pkg update -p -g espressif32` to io `pkg update -g -p espressif32` to make it run.
```console
$ pio pkg update -p -g espressif32
Usage: pio pkg update [OPTIONS]
Try 'pio pkg update -h' for help.

Error: Got unexpected extra argument (espressif32)
```

```console
$ pio pkg update -g -p espressif32
Platform Manager: espressif32@2024.4.12 is already up-to-date
Tool Manager: framework-arduinoespressif32@3.0.0+sha.11140aa is already up-to-date
Tool Manager: tool-esptoolpy@4.7.2 is already up-to-date
Tool Manager: tool-mkfatfs@2.0.1 is already up-to-date
Tool Manager: tool-mklittlefs@3.2.0 is already up-to-date
Tool Manager: tool-ninja@1.7.1 is already up-to-date
Tool Manager: tool-openocd-esp32@2.1100.20220706 is already up-to-date
Tool Manager: tool-riscv32-esp-elf-gdb@12.1.0+20221002 is already up-to-date
Tool Manager: tool-xtensa-esp-elf-gdb@12.1.0+20221002 is already up-to-date
Tool Manager: toolchain-esp32ulp@1.23500.220830 is already up-to-date
Tool Manager: toolchain-riscv32-esp@12.2.0+20230208 is already up-to-date
Tool Manager: toolchain-xtensa-esp32@12.2.0+20230208 is already up-to-date
Tool Manager: toolchain-xtensa-esp32s3@12.2.0+20230208 is already up-to-date
```

Then, after that command, it said that it is up to date but the resulting firmware (clean and build) continued to crash.

So, I just removed all esp-idf folders inside .platformio folder. PIO downloaded all that again and now I have the latest 6.7.0 which works great!

Thanks!!!

ascillato, 2024-05-15:
@valeros

Just to learn, what was the issue? What changed? Can you link to the commit that fixed this? Thanks.

valeros, 2024-05-15:
Our package was missing the patch from based commit https://github.com/espressif/esp-idf/commit/bd56ca48be81763872d1562822b94843eed12fdc.

QuAzI, 2024-06-08:
It looks like build https://registry.platformio.org/platforms/platformio/espressif32 still has this issue as esphome still build wrong firmwares (for ESP32-WROOM-32/nodemcu-32s in my case)

Jason2866, 2024-06-08:
@QuAzI The Platformio registry has latest bug fixed versions.

gojimmypi, 2024-07-13:
Thanks @valeros your https://github.com/platformio/platform-espressif32/issues/1379#issuecomment-2113134037 was tremendously helpful.  And thanks @Jason2866 for pointing me in this direction. I don't know how long it might have otherwise taken me to resolve my `panic'ed (Cache disabled but cached memory region accessed)` error. 

I've finally been able to create [my own custom PlatformIO ESP-IDF release](https://github.com/gojimmypi/esp-idf/releases/tag/v5.2.2.20240714) (yay!) that will include some wolfSSL updates for using the `esp-tls` layer. See also  https://github.com/espressif/esp-idf/issues/13966.