platformio/platform-espressif32 #1386: After upgrading to 6.7.0 - an error!

butaikis, 2024-05-29:
Hi. After upgrading the platform to 6.7.0, I needed to enable Menuconfig = Component config = Log output = Default log verbosity Set Verbose. After compiling and running, I see an error

```
I (2360) heap_init: At 40096A00 len 00009600 (37 KiB): IRAM
I (2365) spiram: Adding pool of 3839K of external SPI memory to heap allocator
D (2372) FLASH_HAL: extra_dummy: 2
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).

Core  0 register dump:
PC      : 0x401247e4  PS      : 0x00060535  A0      : 0x80085752  A1      : 0x3ffe3af0  
A2      : 0x3ffbf068  A3      : 0x3ffe3b84  A4      : 0x001740c8  A5      : 0x3ffe3b40
A6      : 0x3ffe3b20  A7      : 0x0000000c  A8      : 0x80095995  A9      : 0x3ffe3ac0
A10     : 0x00000000  A11     : 0x3ffe3af0  A12     : 0x3ff42000  A13     : 0x4008f310
A14     : 0x00040000  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x401247e1:0x3ffe3af0 0x4008574f:0x3ffe3b30 0x4013b5f5:0x3ffe3b60 0x4008583d:0x3ffe3b80 0x4013b62c:0x3ffe3bb0 0x4011fda8:0x3ffe3c00 0x400832a5:0x3ffe3c40 0x400797d7:0x3ffe3c70 |<-CORRUPTED
```

There was no such thing on the last version of Espressif 32

Jason2866, 2024-05-29:
See #1379

butaikis, 2024-05-29:
@Jason2866 
Thank you. Tell me, how can I enable the same detailed error output as in this [message](https://github.com/platformio/platform-espressif32/issues/1379#issue-2296481325)?

Jason2866, 2024-05-29:
`monitor_filters  = esp32_exception_decoder`

closed 2024-05-29 by butaikis