platformio/platform-espressif32 #1387: After the update, the esp32C3 cannot be used with the USB upload program anymore

jie326513988, 2024-05-31:
```
[env:seeed_xiao_esp32c3]
platform = espressif32
board = seeed_xiao_esp32c3
framework = arduino

monitor_speed = 115200
upload_speed = 2000000
board_build.partitions = huge_app.csv
board_build.f_cpu = 160000000L
board_build.flash_size = 4MB
board_build.f_flash = 80000000L
board_build.flash_mode = dio
board_build.filesystem = littlefs
extra_scripts = ./extra_script.py
monitor_filters = esp32_exception_decoder
board_upload.before_reset = usb_reset

build_flags =
  -D ARDUINO_USB_MODE=1
  -D ARDUINO_USB_CDC_ON_BOOT=1
  -D CORE_DEBUG_LEVEL=1
```
After the Espressf 32 version was updated from 6.4.0 to 6.7.0,
I used USB to upload the program once, and ESP32c3 can no longer use USB,
Using the flash_download_tool_1.9.5 tool also prevents the program from being uploaded or cleared,
Ensure that there are no issues with the hardware and program, and that they have been working normally for a long time in 6.4.0,
Switching back to 6.4.0 also prevents the use of USB upload programs.

[image]
[image]
[image]

Jason2866, 2024-05-31:
Known bug in Arduino Core 2.0.16 Is fixed in Core 2.0.17
You can use with 
```
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.17/esp32-2.0.17.zip
```
until Platformio release the core 2.0.17

jie326513988, 2024-06-01:
> Arduino Core 2.0.16 中的已知错误已在 Core 2.0.17 中修复，您可以使用
> 
> ```
> platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.17/esp32-2.0.17.zip
> ```
> 
> 直到 Platformio 发布核心 2.0.17

我删除了2.0.16，并复制了2.0.17，但它无法修复已经被锁定的ESP32C3吗？

jie326513988, 2024-06-01:
> Known bug in Arduino Core 2.0.16 Is fixed in Core 2.0.17 You can use with
> 
> ```
> platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32/releases/download/2.0.17/esp32-2.0.17.zip
> ```
> 
> until Platformio release the core 2.0.17

I deleted 2.0.16, and copied 2.0.17 over, but it doesn't fix the already locked ESP32C3?

jie326513988, 2024-06-01:
It can be fixed by downloading via serial port.
Connect GPIO8 to high level and enter download mode manually.

closed 2024-06-01 by jie326513988