platformio/platform-espressif32 #1389: Add support for ESP32-S3-WROOM-1-N4R2

nicola-reat, 2024-06-03:
Hi everyone, I would like to ask if anyone has already used an esp32-s3-wroom1-n4r2 with platformio

Jason2866, 2024-06-03:
yes.

nicola-reat, 2024-06-03:
> yes.
@Jason2866 

Okk can you send me your configuration??

Jason2866, 2024-06-03:
this boards json
```
{
  "build": {
    "arduino": {
      "memory_type": "qio_qspi"
    },
    "core": "esp32",
    "extra_flags": [
      "-DBOARD_HAS_PSRAM",
      "-DARDUINO_RUNNING_CORE=1",
      "-DARDUINO_EVENT_RUNNING_CORE=1"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi",
    "bluetooth"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "WROOM-1-N4R2",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.espressif.com",
  "vendor": "espressif"
}
```

Jason2866, 2024-06-03:
Take care to use a matching partition scheme! Every scheme for 4MB flash will work.

nicola-reat, 2024-06-03:
> Take care to use a matching partition scheme! Every scheme for 4MB flash will work.

Yes, thanks, I'll implement it now

nikito7, 2024-08-27:
It works ok with Tasmota