platformio/platform-espressif32 #1392: Debugging not hitting breakpoints

aaron-neal, 2024-06-06:
I am trying to debug an ESP32-S2 board. I have succesfully connected the ESP-PROG board and can perfectly program the ESP32 from VS Code, including chanigng the driver with Zadig.

When i press debug, everything seems to work fine, except no breakpoints get hit in VS Code, it seems as if the debugger connects, but doesn't attach to VScode.

platformio.ini
```
debug_tool = esp-prog
upload_protocol = esp-prog
debug_init_break = tbreak setup
```

Debug Console log
```
Processing dev (board: esp32-s2-saola-1; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
....
....
....
esptool.py v4.5.1
Creating esp32s2 image...
Merged 2 ELF sections
Successfully created esp32s2 image.
========================= [SUCCESS] Took 8.11 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
dev            SUCCESS   00:00:08.107
========================= 1 succeeded in 00:00:08.107 =========================
Reading symbols from C:\Users\...\Documents\GitHub\...\.pio\build\dev\******.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : starting gdb server for esp32s2 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s2] Target halted, PC=0x400DADE2, debug_reason=00000000
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 119 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 369 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s2.flash' size 4096 KB
Info : Using flash bank 'esp32s2.flash' size 4096 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 119 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 369 KB
Info : Using flash bank 'esp32s2.irom' size 372 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 119 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 369 KB
Info : Using flash bank 'esp32s2.drom' size 120 KB
0x40000400 in ?? ()
Loading section .dram0.data, size 0x2a30 lma 0x3ffc0b60
Loading section .iram0.vectors, size 0x403 lma 0x40024000
Loading section .iram0.text, size 0xc758 lma 0x40024404
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40007299, debug_reason=00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1010 ms). Workaround: increase "set remotetimeout" in GDB
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1013 ms). Workaround: increase "set remotetimeout" in GDB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Erased 122880 bytes in 1598.49 ms
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F9, debug_reason=00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1032 ms). Workaround: increase "set remotetimeout" in GDB
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Erased 380928 bytes in 2694.44 ms
Loading section .flash.appdesc, size 0x100 lma 0x3f000020
Loading section .flash.rodata, size 0x1db08 lma 0x3f000120
Loading section .flash.text, size 0x5c5bf lma 0x40080020
Info : PROF: Compressed 121864 bytes to 33137 bytes in 1.999000ms
Info : PROF: Data transferred in 327.764 ms @ 98.7306 KB/s
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Wrote 121864 bytes in 2418.25 ms (data transfer time included)
Info : PROF: Compressed 378303 bytes to 250912 bytes in 11.001000ms
Info : PROF: Data transferred in 2601.17 ms @ 94.2004 KB/s
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Wrote 378303 bytes in 4488.85 ms (data transfer time included)
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Start address 0x400272bc, load size 563026
Transfer rate: 35 KB/sec, 14436 bytes/write.
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
[esp32s2] Debug controller was reset.
[esp32s2] Core was reset.
[esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s2] Core was reset.
[esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Temporary breakpoint 1 at 0x4008afd0: setup. (2 locations)
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak setup`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
```

However, it appears the code isn't halted (watchdog LED is flashing) and no breakpoints are hit, including at setup.

Any help greatly appreciated.

aaron-neal, 2024-06-10:
I will have to check later today, I did however notice the  "2 locations" for the setup function so I renamed one of the functions in a library to see if this was the issue, but still did not work.

I also span up a linux VM to test and I had the same issue as on my Windows install.

I haven't tried with a simple blink sketch yet, but plan to do that as well.

aaron-neal, 2024-06-30:
Hi @valeros, sorry to revisit this, I still have no success.

Just tried a blank project, again everything programs fine, seems like the esp32 is paused, but no breakpoints in VS Code (windows or Linux).

However, I did get this error in vscode (windows): 

[image]

Tried with and without the debug_init_break = tbreak setup.

Anything else I can attempt?

aaron-neal, 2024-06-30:
Also when I press pause, I get this in the teminal:

[image]

valeros, 2024-07-01:
I'm curious what happens if you start a new debug session using CLI? Just open a new terminal window directly in the IDE and run the following command: `pio debug --interface=gdb -x .pioinit`

aaron-neal, 2024-07-01:
Thanks for your time, not seeing any errors and no debug points being hit, but VScode doesn't think it is debugging using this command, is there something I need to do in order to attach to this instance?

Question.... is it right that when hitting debug, the project gets built first in the normal terminal and then again in the debug console tab?

```
Reading symbols from C:\Users\********\Documents\GitHub\***************\.pio\build\dev\************_1.5.2.elf...
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = esp-prog
PlatformIO: Initializing remote target...
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

adapter speed: 5000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 5000 kHz
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s2 on pipe
Info : accepting 'gdb' connection from pipe
Info : [esp32s2] Target halted, PC=0x40014980, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 121 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 376 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s2.flash' size 4096 KB
Info : Using flash bank 'esp32s2.flash' size 4096 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 121 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 376 KB
Info : Using flash bank 'esp32s2.irom' size 380 KB
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 121 KB
Info : Flash mapping 1: 0x30020 -> 0x40080020, 376 KB
Info : Using flash bank 'esp32s2.drom' size 124 KB
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1026 ms). Workaround: increase "set remotetimeout" in GDB
0x40014980 in ?? ()
Error: FreeRTOS maximum used priority is unreasonably big, not proceeding: 106
Loading section .dram0.data, size 0x2a40 lma 0x3ffc0f70
Loading section .iram0.vectors, size 0x403 lma 0x40024000
Loading section .iram0.text, size 0xcb64 lma 0x40024404
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1007 ms). Workaround: increase "set remotetimeout" in GDB
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Erased 126976 bytes in 1624.15 ms
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x4000728B, debug_reason=00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (1018 ms). Workaround: increase "set remotetimeout" in GDB
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s2] Target halted, PC=0x40031E92, debug_reason=00000001
Info : PROF: Erased 389120 bytes in 2723.98 ms
Loading section .flash.appdesc, size 0x100 lma 0x3f000020
Loading section .flash.rodata, size 0x1e3b0 lma 0x3f000120
Loading section .flash.text, size 0x5e283 lma 0x40080020
Info : PROF: Compressed 124080 bytes to 33791 bytes in 2.088000ms
Info : PROF: Data transferred in 331.936 ms @ 99.4138 KB/s
[... 11 lines cut ...]
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Start address 0x4002738c, load size 573658
Transfer rate: 38 KB/sec, 14709 bytes/write.
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
[esp32s2] Debug controller was reset.
[esp32s2] Core was reset.
[esp32s2] Target halted, PC=0x400004F5, debug_reason=00000000
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s2] Core was reset.
[esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Temporary breakpoint 1 at 0x40092f98: file src/main.cpp, line 360.
PlatformIO: Initialization completed
(gdb) PlatformIO: Resume the execution to `debug_init_break = tbreak setup`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.
```

valeros, 2024-07-02:
> but VScode doesn't think it is debugging using this command

The debug in CLI mode works independently from VSCode, I was just interested if there a problem somewhere between PlatformIO Core and IDE. Apparently, VSCode has nothing to do with the problem. 

> Question.... is it right that when hitting debug, the project gets built first in the normal terminal and then again in the debug console tab?

Do you mean it builds your projects twice from scratch? 

Overall, the log seems fine, although I'd probably set a hardware assisted breakpoint instead of a generic one:
```
debug_init_break = thb setup
```

Any chance you could try other hardware setup with a different debug probe?

aaron-neal, 2024-07-25:
I have now spun up a new HW revision which isolates the JTAG completely away from being used elsewhere (before I had to mod the board). I can still program the board perfectly fine with the esp-prog over JTAG.

Still not having much luck, the only other debuggers I have are a JLink Segger Mini Edu, which doesn't seem compatile and a Jeff Probe (black magic) which again doesn't seem compatible.

What would be a low cost debugger I could look out?

Anything else I can try, getting a bit lost now.

aaron-neal, 2024-07-26:
@valeros quick little update, I have installed the ESP-IDF extension in VSCODE and have succesfully used the esp-prog to program and debug my hardware in vscode including breakpoints etc.... I feel this has proven out the HW side of things as being OK.

I have seen a couple of similar issues that point to it possibly be due to the partition table, anything I can do to rule this out.

Another observation I have made, when the debugger loads the code and fails to stop for breakpoints, if i reset the board and stop debugging, my code does not run, almost like it didnt program / or some other error, I am going to put something on the serial outputs to see what's happening there.

aaron-neal, 2024-08-22:
@valeros I have finally, I think, resolved this issue after many hours of going in circles.

I determined that ```-DBUILD_DATE=${UNIX_TIME}``` should be ```-DBUILD_DATE=$UNIX_TIME```

The curly braces, no idea why I was using them, was causing the weird issues.

I can now debug.

closed 2024-08-22 by aaron-neal

NoClue11, 2025-03-11:
Hi, I was fighting with a similar issue with my new esp-prog and a ESP32-Dev-Module. The reason why it was not working for me - stupid enough - was that the ESP-PROG was set to supply 3.3V but I connected the Dev-Module to 5V. So it was working only partially... After changing the jumper to 5V it started to work.