platformio/platform-espressif32 #1393: Inspect fails

hitecSmartHome, 2024-06-07:
Bunch of logs...

```
Retrieved `.pio\\build\\esp-wrover-kit\\lib6ca\\libRtc.a' from cache\r\nLinking .pio\\build\\esp-wrover-kit\\firmware.elf\r\nRetrieving maximum program size .pio\\build\\esp-wrover-kit\\firmware.elf\r\nCalculating size .pio\\build\\esp-wrover-kit\\firmware.elf\r\nChecking size .pio\\build\\esp-wrover-kit\\firmware.elf\r\n text\t data\t bss\t dec\t hex\tfilename\r\n2896994\t 295008\t 41377\t3233379\t 315663\t.pio\\build\\esp-wrover-kit\\firmware.elf\r\nAdvanced Memory Usage is available via \"PlatformIO Home > Project Inspect\"\r\nRAM: [= ] 12.6% (used 41224 bytes from 327680 bytes)\r\nFlash: [==========] 100.6% (used 3165101 bytes from 3145728 bytes)\r\n\n\nError: The program size (3165101 bytes) is greater than maximum allowed (3145728 bytes)\r\n*** [checkprogsize] Explicit exit, status 1\r\n========================= [FAILED] Took 87.55 seconds ========================="
```

Specific error:

`Error: The program size (3165101 bytes) is greater than maximum allowed (3145728 bytes)`

I have tried to set the debug build type to debug and release and `check_skip_packages = yes`

Jason2866, 2024-06-07:
the app partition is too small for your compiled firmware. Use a partition.csv where the app slot is big enough for your firmware. NOT an issue.

hitecSmartHome, 2024-06-07:
Will try it with `huge_app.csv` thank you very much!

closed 2024-06-07 by hitecSmartHome