platformio/platform-espressif32 #1395: esp32-c6-devkitc-1 defines wrong RAM size

QuickJack, 2024-06-08:


valeros, 2024-06-10:
We use the same size as the Arduino IDE https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L356
As a workaround, you can adjust that value in your `platformio.ini`:
```
board_upload.maximum_ram_size = 524288
```

closed 2024-06-10 by valeros

QuickJack, 2024-06-10:
So Arduino IDE is using the wrong RAM size as well. What is the reason behind this decision?

valeros, 2024-06-10:
> What is the reason behind this decision?

Even though technically the SRAM size is 512kb (distributed between IRAM and DRAM), the `maximum_ram_size` field only represents DRAM size (320kb) used for non-constant static data.

QuickJack, 2024-06-10:
Besides the esp32-c6-devkitm-1 there is a esp32-c6-devkitc-1. They have the following documentation pages which both link to the same datasheet for the ESP32-c6:

- https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitm-1/user_guide.html#related-documents
- https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html#related-documents

However, the template for the esp32-c6-devkitc-1 specifies 512k whereas the template for the esp32-c6-devkitm-1 specifies only 320k. Both boards seem to contain the same chip. Why is maximum_ram_size different for these two templates?

valeros, 2024-06-10:
Thanks for pointing out, that's an oversight, fixed in https://github.com/platformio/platform-espressif32/commit/d91a581a0ecc9a3ba5c9b1fc9e0f50221818b7c1