platformio/platform-espressif32 #1397: pio remote run -t upload missing address

jschneekloth, 2024-06-09:
Seeing an issue with `pio remote run` command.  My setup:
- VM that holds my [platform.io](http://platform.io/) project for esp32-c6 dev kit
- Local on my mac, I have the hardware plugged in and am running pio remote agent start

My platform.ini:
```
[env]
build_flags =
    -std=c++17
    -fexceptions

[env:c3]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf
```

Basically, just building a hello world, but when I run `pio remote run -t upload -e c6 -v` I see the following error:
```
......
Successfully created esp32c6 image.
================================================================================================== [SUCCESS] Took 227.77 seconds ==================================================================================================
Uploading firmware remotely
Processing c6 (platform: espressif32; board: esp32-c6-devkitc-1; framework: espidf; build_flags: -std=c++17, -fexceptions)
--------------------------------------------------------------------------------
<lambda>(["upload"], [".pio/build/c6/firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio/build/c6/firmware.bin"])
Auto-detected: /dev/cu.usbmodem57280257101
"/opt/homebrew/Cellar/platformio/6.1.15/libexec/bin/python" "/Users/jschneekloth/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32c6 --port "/dev/cu.usbmodem57280257101" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect .pio/build/c6/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address ".pio/build/c6/firmware.bin" must be a number
*** [upload] Error 2
```

Looks like the address is missing from the command.  I've worked around this via custom upload script, but this is fragile and should work from platform.

Jason2866, 2024-06-09:
The platformio.ini does not match to the board. Probably not the reason for the issue, but with wrong infos finding issues is even more complicated

jschneekloth, 2024-06-09:
Apologies was trying multiple boards and cross crossed the output, here is corrected information:

ini:
```
[env]
build_flags =
    -std=c++17
    -fexceptions

[env:c6]
platform = espressif32
board = esp32-c6-devkitc-1
framework = espidf

[env:c3]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf
```

c3 output:
`pio remote run -t upload -e c3 -v`
results in:
```
...
Successfully created esp32c3 image.
========================================================================================== [SUCCESS] Took 25.71 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
c6             IGNORED
c3             SUCCESS   00:00:25.712
========================================================================================== 1 succeeded in 00:00:25.712 ==========================================================================================
Uploading firmware remotely
Processing c3 (platform: espressif32; board: esp32-c3-devkitm-1; framework: espidf; build_flags: -std=c++17, -fexceptions)
--------------------------------------------------------------------------------
<lambda>(["upload"], [".pio/build/c3/firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio/build/c3/firmware.bin"])
Auto-detected: /dev/cu.usbserial-1410
"/opt/homebrew/Cellar/platformio/6.1.15_1/libexec/bin/python" "/Users/jschneekloth/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32c3 --port "/dev/cu.usbserial-1410" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect .pio/build/c3/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address ".pio/build/c3/firmware.bin" must be a number
*** [upload] Error 2
========================== [FAILED] Took 0.26 seconds ==========================
```

c6 output:
`pio remote run -t upload -e c6 -v`
results in:
```
Successfully created esp32c6 image.
========================================================================================== [SUCCESS] Took 24.38 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
c6             SUCCESS   00:00:24.381
c3             IGNORED
========================================================================================== 1 succeeded in 00:00:24.381 ==========================================================================================
Uploading firmware remotely
Processing c6 (platform: espressif32; board: esp32-c6-devkitc-1; framework: espidf; build_flags: -std=c++17, -fexceptions)
--------------------------------------------------------------------------------
<lambda>(["upload"], [".pio/build/c6/firmware.bin"])
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
BeforeUpload(["upload"], [".pio/build/c6/firmware.bin"])
Auto-detected: /dev/cu.usbmodem57280257101
"/opt/homebrew/Cellar/platformio/6.1.15_1/libexec/bin/python" "/Users/jschneekloth/.platformio/packages/tool-esptoolpy/esptool.py" --chip esp32c6 --port "/dev/cu.usbmodem57280257101" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect .pio/build/c6/firmware.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address ".pio/build/c6/firmware.bin" must be a number
*** [upload] Error 2
========================== [FAILED] Took 0.25 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
c6             FAILED    00:00:00.247
c3             IGNORED
==================== 1 failed, 0 succeeded in 00:00:00.247 ====================
```

jschneekloth, 2024-06-09:
Workaround I am employing currently is to have a custom upload script specifying the address:
custom_upload_c3.py
```
Import('env')

def before_upload(source, target, env):
    firmware_path = str(source[0])
    env.Replace(
        UPLOADCMD="$PYTHONEXE $UPLOADER $UPLOADERFLAGS 0x10000 %s" % firmware_path
    )

env.AddPreAction("upload", before_upload)
```
With updated ini (for example):
```
[env:c3]
platform = espressif32
board = esp32-c3-devkitm-1
framework = espidf
extra_scripts = shared/custom_upload_c3.py
```

In addition to this, I also needed to be explicit about the partitioning (board I think came with rainmaker? flashed and had non-default layout).

closed 2025-04-25 by stale[bot]