platformio/platform-espressif32 #1398: Update to ESP-IDF v5.2.2

IEatCodeDaily, 2024-06-12:
As title say, an update to ESP-IDF v5.2.2 or at least having an option to use ESP-IDF v5.2.2 would be great.
I'm trying to implement WiFI FTM localization and the FTM report API only exist in v5.2.2.
Or is there a way to force PlatformIO to use v5.2.2?

Also, There isn't really a documentation on the [PlatformIO registry](https://registry.platformio.org/tools/platformio/framework-espidf/versions) as to what ESP-IDF version does each framework-espidf version corresponds to.
Being able to easily see what ESP-IDF version each framework-espidf correspond to would be amazing.

Otherwise I'd have to migrate to ESP-IDF (which I really don't want to. I'm spoiled by PlatformIO at this point).

Thanks!

sivar2311, 2024-06-12:
> Also, There isn't really a documentation on the [PlatformIO registry](https://registry.platformio.org/tools/platformio/framework-espidf/versions) as to what ESP-IDF version does each framework-espidf version corresponds to.

See https://github.com/platformio/platform-espressif32/releases and note the "Supported Frameworks" section

IEatCodeDaily, 2024-06-12:
apparently I can just override the entire /components and /tools folders with no problem (so far).
finger crossed there won't be any hidden problems.

closed 2024-06-18 by valeros

ascillato, 2024-06-21:
@IEatCodeDaily 

As Valeros has already added this to the platformio repository, you can use the new SDK in the actual platform-espressif32 (until the new one is released) by the following in platformio.ini

```ini
[env]
platform                                = espressif32@6.7.0                 ; Includes ESP-IDF v5.2.1
platform_packages                       = framework-espidf @ ~3.50202       ; Forces to use ESP-IDF v5.2.2 compiled by Platformio Group.
framework                               = espidf

```

gojimmypi, 2024-06-30:
Is it possible to use a `git clone` source of an Espressif ESP-IDF? I was hoping to use the very latest master version, but even when trying a [v5.2.2 release](https://github.com/espressif/esp-idf/releases/tag/v5.2.2), I've not had much luck.

I've tried doing something like this:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = espidf
monitor_speed = 115200
platform_packages = 
    framework-espidf @ file://C:/SysGCC/esp32-master/esp-idf/v5.2-master
```

but see errors like this:

```
Processing your_environment (platform: espressif32; board: esp32dev; framework: espidf)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 1.1.0
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp-elf @ 13.2.0+20230928
TypeError: can only concatenate str (not "NoneType") to str:
  File "C:\Users\gojimmypi\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\gojimmypi\.platformio\platforms\espressif32\builder\main.py", line 368:
    target_elf = env.BuildProgram()
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gojimmypi\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gojimmypi\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 121:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Util\envs.py", line 251:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gojimmypi\.platformio\penv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 347:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 676:
    return method(*args, **kw)
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\gojimmypi\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\gojimmypi\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1314:
    ensure_python_venv_available()
  File "C:\Users\gojimmypi\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1287:
    venv_dir = get_idf_venv_dir()
  File "C:\Users\gojimmypi\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1241:
    env.subst("$PROJECT_CORE_DIR"), "penv", ".espidf-" + idf_version
```

I can otherwise successfully compile with PlatformIO releases:
```
[env:esp32dev]
platform = espressif32@6.3.0
board = esp32dev
framework = espidf
monitor_speed = 115200 
```

Am I using the `platform_packages = framework-espidf @ file ...` correctly?

Jason2866, 2024-06-30:
@gojimmypi It is not possible to use the espressif IDF repo as source.

gojimmypi, 2024-06-30:
> It is not possible to use the espressif IDF repo as source nor a file or directory

@Jason2866 thanks for saving me the additional effort of trying to get that to work!

As noted in https://github.com/platformio/platformio-core/issues/4930#issuecomment-2198547206 I'd like to use some changes I'm making to the ESP-IDF in PlatformIO. Are there any alternatives for me to do this, other than a brute-force method of copying files to my `C:\Users\gojimmypi\.platformio\packages\framework-espidf\components\` directory?

Jason2866, 2024-06-30:
@gojimmypi Fork the IDF version you want to use. Do your changes. Add a valid package.json and version.txt in the root. Create a release. The github function release will NOT work since it does not include the git submodules. The generated release can be used with `platform_packages`
Example can be seen here https://github.com/Jason2866/esp-idf and working Platformio releases https://github.com/Jason2866/esp-idf/releases

gojimmypi, 2024-06-30:
> Fork the IDF version you want to use. Do your changes. Add a valid package.json and version.txt in the root. Create a release.

@Jason2866 that sounds like an excellent plan! Thank you so much. I'll give that a try! :)

> I don't like mbedtls at all. Would like to try wolfssl 

I hear ya on the mbedTLS topic. It's clearly not a priority for them, but it is for wolfSSL.

I have a copy of the Espressif HTTP/S example [here](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix/IDE/Espressif/ESP-IDF/examples/esp_http_client_example) with the ESP-IDF updates [on my wolfssl branch](https://github.com/gojimmypi/esp-idf/tree/wolfssl/components/esp-tls). It basically works, minus the "certificate bundle" feature that is specific the mbedTLS at this time.  See also: https://www.wolfssl.com/forums/post7724.html

I do plan on resolving the certificate bundle topic, but for now I believe wolfSSL is otherwise working on my ESP-IDF fork with the `esp-tls` layer. Clearly it still needs a bit of cleanup and testing before creating a PR.

There are more wolfSSL examples here (not using the `esp-tls` layer): https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples

> Could you ping me if your wolfssl implementation is ready to use?

Absolutely! Do you have a specific project in mind?  Note that wolfSSL simply for the PlatformIO environment should be working just fine:

https://www.wolfssl.com/wolfssl-now-supported-on-platformio/

The issue I am working on is for those that _really_ want to use the `esp-tls` layer. It's not something I would recommend, as it only adds a layer of complexity. But I understand how some might want it - particularly in an early evaluation stage to easily use either from the ESP-IDF.

If you'd like to take this for a test drive, let me know. I'd love to have your input.

Feel free to reach out to support@wolfssl.com as well.

Cheers

commit 64608ba2 (2024-06-18): Update ESP-IDF to v5.2.2

Resolves #1398