platformio/platform-espressif32 #1400: The board selection list has many duplicate items.

Tinyu-Zhao, 2024-06-14:
As shown in the figure, there are many duplicate items in the board selection list.
[image]

valeros, 2024-06-17:
Hi @Tinyu-Zhao, any chance you have multiple `espressif32` platforms installed with different versions? What's the output of the `pio pkg list -g --only-platforms` command?

Tinyu-Zhao, 2024-06-18:
> Hi @Tinyu-Zhao, any chance you have multiple `espressif32` platforms installed with different versions? What's the output of the `pio pkg list -g --only-platforms` command?

[image]

Yes, sometimes the problem with customer feedback is the different versions of the framework that I often need to change.

closed 2025-02-25 by Tinyu-Zhao