platformio/platform-espressif32 #1407: A crash of the application when upgrading from 6.6.0 to 6.7.0 using esp-idf

grisharav, 2024-06-23:
After upgrading the project from:
```
framework = espidf
platform = espressif32@6.6.0
``` 
to
```
framework = espidf
platform = espressif32@6.7.0
```
We have a crash when calling `esp_core_dump_init()` here's a partial stacktrace:

```
  #0  0x401e2491 in esp_cpu_clear_watchpoint at ??:?
  #1  0x3ffb3890 in _xt_exception_table at ??:?
  #2  0x4008aecd in vPortEnterCritical at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa\include\freertos/portmacro.h:571
      (inlined by) prvTaskEnterCriticalSafeSMPOnly at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\freertos\esp_additions\include/esp_private/freertos_idf_additions_priv.h:68
      (inlined by) xTaskIncrementTick at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel/tasks.c:3194
  #3  0x3ffb38c0 in _xt_exception_table at ??:?
  #4  0x4008d303 in xPortSysTickHandler at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\freertos/port_systick.c:217
  #5  0x3ffb38e0 in _xt_exception_table at ??:?
  #6  0x4008d095 in _frxt_timer_int at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/portasm.S:344
  #7  0x3ffb3900 in _xt_exception_table at ??:?
  #8  0x400875d2 in _xt_lowint1 at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\xtensa/xtensa_vectors.S:1240
  #9  0x3ffb3910 in _xt_exception_table at ??:?
  #10 0x4008ddee in spi_flash_ll_get_addr_bitlen at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\hal\esp32\include/hal/spi_flash_ll.h:336
      (inlined by) spi_flash_hal_read at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\hal/spi_flash_hal_common.inc:188
  #11 0x4009515a in spi_flash_chip_generic_read at C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/spi_flash/spi_flash_chip_generic.c:246
  #12 0x40084413 in esp_flash_read at C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/spi_flash/esp_flash_api.c:899
  #13 0x40119257 in esp_partition_read at C:\Users\GregRavikovich\.platformio\packages\framework-espidf\components\esp_partition/partition_target.c:38
  #14 0x4011c3c0 in esp_core_dump_image_check at C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/espcoredump/src/core_dump_flash.c:388
  #15 0x4011c436 in esp_core_dump_init at C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/espcoredump/src/core_dump_flash.c:345
```

I am not quire sure what were the changes between the releases that could cause something like this. Can someone point me in the right direction?

toine512, 2024-06-23:
You don't show the actual exception.
My app doesn't even boot while ESP-IDF version is supposed to have not changed. Something's wrong.

grisharav, 2024-06-24:
Something is strange indeed. I have removed all the global packages and frameworks cache, cleared all libraries folder, rebuilt the software, and the crash is gone, behaves properly now.

closed 2024-07-09 by valeros