platformio/platform-espressif32 #1411: Can't build for esp32-S3 : Cannot find binary 'xtensa-esp32s3-elf-ranlib'

MaximeM, 2024-06-29:
With the default arduino project, no modification, tried lot of different board... but no always this error !
```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Archiving .pio\build\esp32-s3-devkitc-1\libFrameworkArduino.a
Indexing .pio\build\esp32-s3-devkitc-1\libFrameworkArduino.a
xtensa-esp32s3-elf-gcc-ranlib: Cannot find binary 'xtensa-esp32s3-elf-ranlib'
*** [.pio\build\esp32-s3-devkitc-1\libFrameworkArduino.a] Error 1
```

Jason2866, 2024-06-29:
Probably od stuff left which does disturb. Close VSC (if you use). Delete the hidden folder `.platformio` with all content. Platformio will rebuild and fetch all needed from the Internet.

MaximeM, 2024-06-29:
did close/ made mutiple project...

Just deleted the product, same happened...

Turns out Avast is putting  xtensa in quarantine.

closed 2024-06-29 by MaximeM