platformio/platform-espressif32 #1417: Request to update ESPTool package

nullstalgia, 2024-07-05:
The last time `esptool.py` supplied with PlatformIO recieved an update was over a year ago. Since then, there have been a lot of improvements, especially in regards to reliability of starting and maintaining connections.

I've been using the ESP32-S3 in some recent projects, and I've found that the chips native USB-Serial port works great for flashing and debugging, but works *way* more reliably with those newer releases.

Note: There has been some slight changes in dependencies, so it's not a simple `git pull` and done. But I was able to get flashing with only a single `pip install IntelHex` on the platformio pyenv.

dlbogdan, 2024-08-27:
see https://github.com/platformio/platform-espressif32/issues/1225

MichaelEischer, 2025-04-29:
In case someone needs this (e.g. to flash an esp32c6), you can use the following snippets to automatically setup esptool 4.8.1:

Add the `platform_packages` and `extra_scripts` to your `platformio.ini`.
```
[env:seeed_xiao_esp32c6]
platform = https://github.com/platformio/platform-espressif32.git
board = seeed_xiao_esp32c6
framework = espidf
platform_packages =
    tool-esptoolpy@~1.40801.0
extra_scripts = extra_deps.py
```

Create `extra_deps.py` with the following content.
```python
Import("env")

# Install missed package
try:
    import intelhex
except ImportError:
    env.Execute("$PYTHONEXE -m pip install intelhex")
```

grisharav, 2025-08-03:
I have an ESP32 IDF project which I now upgraded to latest `platform = espressif32@6.12.0` and I got (during compilation):
```
ModuleNotFoundError: No module named 'intelhex'
```
Added
```
platform_packages = tool-esptoolpy @ https://github.com/pioarduino/esptool/releases/download/v4.8.11/esptool.zip
```
Seemed to work, but when I wanted to upgrade to latest 5.0.2, I got:
```
ModuleNotFoundError: No module named 'rich_click'
```
Any ideas what's the issue?

Jason2866, 2025-08-03:
See the post above from Valeros

valeros, 2025-08-04:
@grisharav It should work out of the box. Could you share here a larger portion of the log (preferably in verbose mode) so we can see exactly where the issue occurred?

grisharav, 2025-08-04:
> [@grisharav](https://github.com/grisharav) It should work out of the box. Could you share here a larger portion of the log (preferably in verbose mode) so we can see exactly where the issue occurred?

Sure, without the line `platform_packages = tool-esptoolpy@https://github.com/pioarduino/esptool/releases/download/v4.8.11/esptool.zip`:
```
Traceback (most recent call last):
  File "/home/greg/.platformio/packages/tool-esptoolpy/esptool.py", line 41, in <module>
    import esptool
  File "/home/greg/.platformio/packages/tool-esptoolpy/esptool/__init__.py", line 42, in <module>
Compiling .pio/build/gpt_rev2/libad6/sgtcm_lib/esp32/network/esp32_wifi_mesh_network_manager.o
    from esptool.bin_image import intel_hex_to_bin
  File "/home/greg/.platformio/packages/tool-esptoolpy/esptool/bin_image.py", line 16, in <module>
    from intelhex import HexRecordError, IntelHex
ModuleNotFoundError: No module named 'intelhex'
*** [.pio/build/gpt_rev2/bootloader.bin] Error 1
```

With the line compiles just fine.

valeros, 2025-08-04:
@grisharav can you share here also the header of the log?

valeros, 2025-08-04:
@Jason2866 I understand, but @grisharav is trying to use the default v4 package that should work as expected.

Jason2866, 2025-08-04:
@grisharav just do not add manually a esptool version via platform_packages. Latest Platform will use an actual esptool v4.x version

grisharav, 2025-08-05:
> [@grisharav](https://github.com/grisharav) Could you please try the latest `esptoolpy` package? Just run `pio pkg update -g -t tool-esptoolpy` if it's already installed.

```
pio pkg update -g -t tool-esptoolpy
Tool Manager: tool-esptoolpy@4.8.11 is already up-to-date
```

grisharav, 2025-08-05:
> That's strange. OK, can you remove this folder `/home/YOUR_USER/.platformio/packages/tool-esptoolpy` and then run your project again?

Indeed compiled fine on my machine after the removal. What could be the root cause for this?

valeros, 2025-08-06:
@grisharav 

> What could be the root cause for this?

Hard to guess, probably a mix up of different `esptoolpy` packages.