platformio/platform-espressif32 #1419: ESP PROG IS BROKEN

isfandyar01, 2024-07-06:
PlatformIO Core.

Configuration
Operating system: WIN 10 64bit

PlatformIO Version : 6.1.15

Description of problem
ESP-Prog is not working with ESP32 DOIT Kit v1 using PlatformIO. The issue seems to be with the OpenOCD command called by PlatformIO. I was able to run it fine using the Espressif IDE, which uses the following commands for debugging and flashing:

For debugging
-s ${openocd_path}/share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg

For flashing
-s ${openocd_path}/share/openocd/scripts -c 'set ESP32_FLASH_VOLTAGE 3.3' -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg

Steps to Reproduce
Connect ESP32 DOIT Kit v1 with ESP-Prog.
Attempt to debug or flash using PlatformIO.

The content of platformio.ini:

[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
monitor_speed = 115200
build_type = debug
monitor_filters = esp32_exception_decoder
debug_tool = esp-prog
upload_protocol = esp-prog
upload_port = COM10 ; esp prog port

main.cpp
``` #include "driver/gpio.h"
#include "esp_system.h"
#include "esp_sysview_trace.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"
#include <esp_pthread.h>
#include <stdio.h>

extern "C"
{
    void app_main(void);
}

void hello_task(void *pvParameter)
{
    while (1)
    {
        printf("Hello world!\n");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void hello_task_2(void *pvParameter)
{
    while (1)
    {
        printf("************YOU*********!\n");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

#define BLINK_GPIO GPIO_NUM_2

void app_main()
{
    //     esp_sysview_flush(ESP_APPTRACE_TMO_INFINITE);
    xTaskCreate(&hello_task_2, "hello_task_2", 2048, NULL, 0, NULL);
    xTaskCreate(&hello_task, "hello_task", 2048, NULL, 0, NULL);

    esp_rom_gpio_pad_select_gpio(BLINK_GPIO);

    gpio_set_direction(BLINK_GPIO, GPIO_MODE_OUTPUT);

    while (1)
    {
        gpio_set_level(BLINK_GPIO, 1);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        gpio_set_level(BLINK_GPIO, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}```

terminal output

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.4.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50101.230828 (5.1.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Retrieving maximum program size .pio\build\esp32doit-devkit-v1\firmware.elf
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   5.2% (used 16892 bytes from 327680 bytes)
Flash: [==        ]  21.7% (used 227557 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esp-prog
Uploading .pio\build\esp32doit-devkit-v1\firmware.bin
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 20000 kHz

WARNING: boards/esp-wroom-32.cfg is deprecated, and may be removed in a future release.
adapter speed: 5000 kHz

Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu0 examination failed
Error: Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu1 examination failed
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unexpected OCD_ID = ffffffff
Error: Unexpected OCD_ID = ffffffff
Error: Target not examined yet
Error: Target not examined yet
Error: Unexpected OCD_ID = ffffffff
embedded:startup.tcl:1184: Error: ** Unable to reset target **
in procedure 'program_esp' 
in procedure 'program_error' called at file "C:/Users/ISFAND_PC/.platformio/packages/tool-openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 104
at file "embedded:startup.tcl", line 1184
*** [upload] Error 1

valeros, 2024-07-08:
Hi @isfandyar01, unfortunately, we don't have `esp-prog` at hand so we cannot debug the issue for you. According to your logs, OpenOCD is not able to communicate with the debug probe, although it's strange that it works just fine with IDF. What OpenOCD version is used in the IDF development environment? Please also make sure that your application doesn't interfere with the JTAG pins.

isfandyar01, 2024-07-10:
sorry for late reply thing is i believe platformio used patched version of open ocd with esp-idf framework
i was able to get same setup working with consistency using esp-ide based on eclipse which is provided by espressif
in past the setup was working very well with platformio and it was very stable but latest platformio built have broken it

closed 2025-04-25 by stale[bot]