platformio/platform-espressif32 #1424: SD_MMC sector size not found

ra0943-VK3ACH, 2024-07-14:
When using platformio in vscode SD_MMC does not have the member numSectors or sectorSize.
```
#include <USB.h>
#include <USBMSC.h>
#include <SD_MMC.h>

void setup(){
Serial.begin(115200);
msc.begin(SD_MMC.numSectors(), SD_MMC.sectorSize());
Serial.printf("Sector: %d\tCount: %d\n", SD_MMC.sectorSize(), SD_MMC.numSectors());
}
```
certain parts of my code have been omitted as they arent relevant to the bug(its a basic USB msc stuff)
This code works in arduino ide just not platformio.
i have tried with platformio.ini
```
[env:development]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32s3usbotg
framework = arduino
```
and 
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
```

Jason2866, 2024-07-15:
This is not working anymore
```
[env:development]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32s3usbotg
framework = arduino
``` 
Platformio does not support Arduino Core 3.0.x  See #1225

valeros, 2024-07-15:
Hi @ra0943-VK3ACH, the API you're referring to is implemented in Arduino core 3.0. See the progress of adding this version here #1225

closed 2024-07-15 by valeros