platformio/platform-espressif32 #1425: Espressif esp32-s3-devkit-c1

wolfino65, 2024-07-15:
Maybe I didn't notice it but i do not see the board that is the description or at least the one listed did not work.

valeros, 2024-07-15:
What's the error message exactly?

wolfino65, 2024-07-15:
The thing is there are no error messages, upload is succesfull, but nothing happens, but when I use the arduino IDE, same code, same cable, same esp it works.

sivar2311, 2024-07-15:
> but when I use the arduino IDE, same code, same cable, same esp it works.

Accordingly, your settings in the ArduinoIDE and the platformio.ini are different!
Show a screenshot of the ArduinoIDE setting, the content of your platformio.ini and your sketch.

wolfino65, 2024-07-16:
[image]
[image]
[image]
The esp is currently not plugged in. In the ini file the env did not change since i changed the board in the project (with a fresh brain i realise this might be the problem). Other than that small litle text on the bottom this is the other tah i found. 
[image]
Sorry if its not the settings you were looking for i just recently started using esp and arduino altogether.
P.s.: Thank you for your help @sivar2311

sivar2311, 2024-07-16:
That's strange! The builtin RGB LED is on GPIO 48 per 

https://github.com/espressif/arduino-esp32/blob/6bd3274d3b79bd576b4660b73129856a2cba4310/variants/esp32s3/pins_arduino.h#L13 

The following sketch and platformio.ini works perfectly on my esp32-s3-devkitc-1:

`main.cpp`:
```C++
#include <Arduino.h>

void setup() {
    Serial.begin(115200);
}

void loop() {
    Serial.println("red");
    neopixelWrite(RGB_BUILTIN, 255, 0, 0);
    delay(1000);

    Serial.println("green");
    neopixelWrite(RGB_BUILTIN, 0, 255, 0);
    delay(1000);

    Serial.println("blue");
    neopixelWrite(RGB_BUILTIN, 0, 0, 255);
    delay(1000);
}
```

`platformio.ini`:
```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
monitor_speed = 115200
```

Note: The board is connected to the "COM" labeled USB port (UART).

wolfino65, 2024-07-16:
My esp is the 1.1 version that is why its on pin38. Can the problem be related to the monitor speed be related or i have seen upload speed defined somewhere. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

sivar2311, 2024-07-16:
> Can the problem be related to the monitor speed

As your sketch has no serial output, this cannot be the cause.

Which Arduino core version do you use in the ArduinoIDE and which version in PlatformIO?

2.0.16. is the latest version available for PlatformIO.
There is a problem that the ESP32 does not restart automatically when uploading via HWCDC. Compare both version numbers. Try a manual restart of the ESP32 and see if your sketch runs.

wolfino65, 2024-07-16:
[image]
I'm guessing these are the versions you are talking about.

sivar2311, 2024-07-16:
No, i am talking about the ESP32 Arduino Core versions.

When building in PlatformIO: Check the output:
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 ```
Here 2.0.16 is used

In ArduinoIDE use the Boards Manager and filter for ESP32:
[image]

Here 2.0.14 is installed

wolfino65, 2024-07-16:
[image]
This is the only difference, but i it works from the arduino IDE.

sivar2311, 2024-07-16:
Arduino 3.x ist NOT available to PlatformIO (yet).
Latest available version is 2.0.16.

There is currently a lively debate about this - see #1225

Have you tried the code without the Adafruit Neopixel library?
A simple `neopixelWrite` is enought to drive the builtin LED.

wolfino65, 2024-07-16:
I have not, but i assume it would change nothing simply because not even serial was working right. Everything was working fine in the arduino IDE.
Thank you for your time and help.

sivar2311, 2024-07-16:
> I have not, but i assume it would change nothing simply because not even serial was working right. Everything was working fine in the arduino IDE.
> Thank you for your time and help.

There are two USB ports: COM (UART) and USB (native USB of the S3). Which one do you use?

The Serial object is controlled by `ARDUINO_USB_CDC_ON_BOOT` 
If this is set to 1 (enabled) the output of the Serial object is routed to the native USB port.

See https://community.platformio.org/t/esp32-s3-native-usb-interface-and-serial-monitor-missing-first-messages/40377/9?u=sivar2311