platformio/platform-espressif32 #1427: 6.5.0 -> 6.6.0: 'MSPI_TIMING_FLASH_DEFAULT_CONFIG_ID_CORE_CLK_240M_MODULE_CLK_80M_STR_MODE' undeclared

0xPIT, 2024-07-18:
Migrating from 6.5.0 to 6.6.0, I see the following compile error:

```
~/.platformio/packages/framework-espidf@3.50201.240515/components/esp_hw_support/port/esp32s3/mspi_timing_tuning_configs.h:180:54: error: 

'MSPI_TIMING_FLASH_DEFAULT_CONFIG_ID_CORE_CLK_240M_MODULE_CLK_80M_STR_MODE' undeclared (first use in this function); did you mean 'MSPI_TIMING_FLASH_DEFAULT_CONFIG_ID_CORE_CLK_240M_MODULE_CLK_120M_STR_MODE'?
```

Is this a bug in this component?
Any hint on how to fix this?

valeros, 2024-07-18:
Hi @0xPIT, the issue tracker in this repository is not the right place for generic questions related to any framework. FYI, In v6.6.0 the IDF package was updated from 5.1 to 5.2. Apparently, there was a breaking change between those versions, so I'd recommend reading the release notes for that IDF v5.2.

closed 2024-07-18 by valeros

0xPIT, 2024-07-18:
ok, thanks, I'll try at the esp-idf tracker.